raspberrypi/picamera2 #509: [HOW-TO] Trigger a recording from a web server page where the stream is viewable

azsde, 2023-01-10:
Hello,

I have a use case where I need to be able to view the stream of the camera on a web page, and also be able to manually start/stop the recording when I want to and still be able to view the stream.

[image]

Would it be possible to do so ? If so, how to proceed ?

I thank you in advance for your answers.

Regards,

Azsde

davidplowman, 2023-01-10:
Hi, thanks for the question.

Currently Picamera2 only lets you run one encoder with a video stream, so this sounds doable if you're happy to record an MJPEG file, and serve the same MJPEG frames to the web client, but not so easy if you wanted to record a different format. 

There's an [MJPEG server example](https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server.py) which I assume you've found. In your case you possibly want two outputs, one to the network and one to a file.

Creating an encoder with two outputs is described in section 9.3 of the Picamera2 manual, the only catch is that I don't think you can start/stop the outputs independently. Possibly you could get round this by deriving your own output type and defining a `_write` method that lets you do this?

@chrisruk may have further advice.

azsde, 2023-01-10:
Hi and thanks for your answer :)

I wouldn't mind having an mjpeg video as a result, I could always re-encode it later on if needed.

From your answer, I assume it would be possible to switch from one output (stream) to two (stream + record), the only caveat being that the stream would be interrupted. Is it correct ?

That could be a good 1st approach to my needs, but creating my own output type to avoid having stream interruption would be best obviously.

davidplowman, 2023-01-11:
In the first instance, I would make the simplest UDP example that you can and delete anything to do with a second output. Does that work? If not, what does VLC report or do? Are you able to check whether packets are actually arriving?

Also, and maybe you're aware of this already, streaming h.264 to a web page (rather than VLC) is more difficult than you might imagine, which is why people still end up using MJPEG quite a lot!

azsde, 2023-01-11:
Sorry about that, for some reason, VLC on windows require ffmpeg command to specify the packet size, otherwise it will only display a blank screen.

So, for a VLC windows client, a command that seems to work is : `output1 = FfmpegOutput("-f mpegts udp://<ip>:<port>?pkt_size=1316")`

I've found this value of 1316 on this forum: https://forum.videolan.org/viewtopic.php?t=76524

For a Linux based VLC client, the command specified in the documentation works out of the box, without specifying a packet size.

azsde, 2023-01-20:
Hello @davidplowman 

On a RPI 3, I have really poor performances when I try to stream 1080p@30 fps via a MJPEG server.

When I stream an H264 video feed however, it works flawlessly.

I assume this is due to the fact that MJPEG is encoded using the CPU (SW), but H264 is encoded using the GPU ? (HW)

davidplowman, 2023-01-20:
Hi, you should be able to use a h/w encoder too, for example: https://github.com/raspberrypi/picamera2/blob/next/examples/mjpeg_server_2.py

azsde, 2023-01-20:
It works flawlessly using your example, thank you.

Would you say there is a difference in quality of the recorded / streamed video using MJPEGEncoder vs H264Encoder ?

My goal is to achieve the maximum possible quality when recording the video, but not necessarely for the streamed video, I just need to be able to monitor the camera feed and trigger manual recording, but as I said, I need the recorded video to be recorded at the best quality possible.

davidplowman, 2023-01-20:
H264 will certainly give you much better quality for the same bitrate. But JPEG will look pretty good if you throw enough bits at it, available bandwidth may be a factor. It may also depend on what your streaming client is - the seemingly simple task of streaming H264 to a web page is still, after all these years, absurdly difficult.

azsde, 2023-01-20:
I have tried looking at other options, such as using libvlc and creating a custom Android app to achieve my goal, but so far I am having weird issues with VLC, I have opened a topic on their forums: https://forum.videolan.org/viewtopic.php?f=2&t=161709
Have you experienced such issues on your side ?

davidplowman, 2023-01-20:
We're certainly aware of problems playing back h.264 streams using vlc. Current versions show this nasty blocky behaviour, though older versions were fine. I think we've noted this on our documentation page, but the whole thing is unfortunately rather out of our control.

azsde, 2023-01-20:
@davidplowman thank you for letting me know, I'll try out with older VLC versions.

I did indeed find the warning on this page: https://www.raspberrypi.com/documentation/computers/camera_software.html#rtsp

It's a surprising issue given how popular VLC is.

azsde, 2023-01-23:
@davidplowman FYI the [mjpeg_server_2.py](https://github.com/raspberrypi/picamera2/blob/next/examples/mjpeg_server_2.py) example you provided doesn't work with security camera softwares (I tried Synology Surveillance Station & MotionEye).

For some reason, those software assume that the stream will be available at :

`http://<ip-of-the-pi>:<port>/stream.mjpeg`

But you example put the stream to:

`http://<ip-of-the-pi>:<port>/stream.mjpg`

Notice the subtle difference, I didn't understand why it didn't work and was going crazy until I found out that mjpeg and mjpg are not the same :man_facepalming: 

It might be worth updating the examples so that they can be used right out of the box :)

azsde, 2023-01-24:
On this specific ticket I think it can be closed, however I was planning on posting a link to a working example achieving what I want so that it could help other people.

If you'd rather see this ticket closed go ahead as I don't know when I will have the time to do so.

davidplowman, 2023-01-24:
Thanks, I think I'll close this one for now. Please do open another ticket if something else comes up!

closed 2023-01-24 by davidplowman