raspberrypi/picamera2 #518: [SUGGESTION] Inline SPS headers for h264

RolandasRazma, 2023-01-13:
**Please only include one suggestion or features request per issue!**

picamera v1 was able to inject SPS headers to h264 (MMAL_PARAMETER_VIDEO_ENCODE_SPS_TIMING) would be nice to have in picamera v2 H264Encoder as well

davidplowman, 2023-01-16:
Hi, thanks for the question. Please try setting the codec's `repeat` parameter (section 7.1.1 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf)). Does that do what you need? Thanks!

RolandasRazma, 2023-01-16:
Hi,
I do have as in `... = H264Encoder(repeat=True, iperiod=30)` but it doesn't work in my case. Could it be because I'm using it with `CircularOutput` and copying `CircularOutput._circular` on trigger?

```python
    _encoder = H264Encoder(repeat=True)
    _encoder.output = CircularOutput(buffersize=..., outputtofile=False)
    _camera.start_recording(encoder=_encoder, output=_encoder.output)

    def stream_copy_to(output: BytesIO):
        with _encoder.output._lock:
            circular = _encoder.output._circular.copy()
            got_key_frame = False

            while circular:
                frame, keyframe = circular.popleft()
                if keyframe:
                    got_key_frame = True

                if got_key_frame:
                    output.write(frame)

    # on trigger
    stream = BytesIO()
    stream_copy_to(stream)

    _encoder.output._circular.clear()
    _encoder.output._firstframe = True
```

RolandasRazma, 2023-01-16:
30fps is definitely default value (no header), but I do see correct FPS with picamera1  with `._camera.start_recording(_stream, format='h264', level='4.2', sps_timing=True)`

davidplowman, 2023-01-16:
A bit of investigation suggests the the V4L2 API does actually allow this using the VIDIOC_S_PARM ioctl. It should be possible to call this to set the "time per frame". It looks to me as though all the necessary C types are defined in the Python V4L2 module.

Picamera2 doesn't, by itself, know what framerate you're using as they can be variable, so it might be the kind of thing an application would have to do explicitly if it wanted a particular value to be recorded.

RolandasRazma, 2023-01-16:
thanks for quick replies, definitely not expected.
What about `controls.FrameDurationLimits` and `controls.FrameRate`?

davidplowman, 2023-01-18:
Those controls do give you control over the camera framerate (although it doesn't need to be a single fixed value), but the problem with the h.264 stream is that it has nowhere to record actual frame timestamps. You can put a time interval into the header (such as 30fps), but nothing forces it to be correct  (for example, if frames get dropped somewhere).

Out of curiosity, what are you using to play back these streams that respects this frame interval in the header?

RolandasRazma, 2023-01-18:
streams go to server and put into mp4 container. Where they played on the web at later time. 
At the moment h264 headers set fps on mp4 and when played back on a web it plays back at correct speed (more or less). 
Without headers (or manually specifying fps at conversion time) videos playback at wrong speed.
I guess I could send `controls.FrameRate` as well to server, but with picamerav1 was no need...

davidplowman, 2023-01-18:
Generally I'm a bit nervous about the timing parameter in the h.264 headers, and suspect that creating containers with proper timestamps is better. Picamera2 gives you a few options to help, such as outputting accurate timestamp files, or even muxing stuff straight into an mp4 (if you don't mind it running ffmpeg in the background to do that).

Having said all that, there is a PR [here](https://github.com/raspberrypi/picamera2/pull/529) that allows an (approximate) framerate to be put into the SPS headers. Are you able to give it a try and see if it works for you?

davidplowman, 2023-03-09:
Closing because this feature should be in release 0.3.9.

closed 2023-03-09 by davidplowman