raspberrypi/picamera2 #520: [HOW-TO]Is there a way to disable main?

tf4ong, 2023-01-15:
Is there a way to disable main stream? I want to get only raw stream data.

Will disable main stream increase performance? That is will there be less processes running?

davidplowman, 2023-01-16:
Hi, thanks for the question. I'm afraid it's not possible to disable the "main" stream. In fact, for our libcamera implementation we have to produce a non-raw output in order to drive the camera, so we'd be doing it internally even if Picamera2 let you disable it! You can reduce any performance effects by asking for a small output size, such as QVGA.

davidplowman, 2023-01-18:
Hi again, I don't there's anything more to add to this one, so I'll close it shortly if there's nothing further. Thanks!

davidplowman, 2023-01-20:
Closing, but please open another issue if anything else comes up. Thanks!

closed 2023-01-20 by davidplowman