raspberrypi/picamera2 #521: [BUG] THe framerate for picamera v1 is slower than on picamera module 1

tf4ong, 2023-01-15:
On picamera module from wave share (the legacy one), the framerate for picamera v1 can go up to 90 fps at 640x480 for h264 video recording
However, the maximum speed in picamera module 2 is only at 60 fps.

Is there a reason for this decreases in speed?

davidplowman, 2023-01-16:
Picamera2 (and libcamera underneath it) use open source camera drivers, and these may implement different resolutions and/or framerates from the proprietary firmware that was used previously. In this case I assume you're talking about the ov5647 - the open source driver appears only to implement the 480p60 mode and not the 480p90. Bit I don't know why.

RolandasRazma, 2023-01-16:
is this the same reason for resolutions and crops difference? was using `1280x720` and was surprised to see different crop on picamera v2

davidplowman, 2023-01-16:
You should be able to control exactly which camera mode gets chosen (and hence what field of view you get) by setting the raw stream mode. Please check the documentation on "Raw streams" in section 4.2.2 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf).

davidplowman, 2023-01-20:
Hi, I think I'm going to propose that we close this as there's nothing we can do in Picamera2 to add new sensor modes to a kernel driver. One could file an issue about the missing mode against the Pi Linux kernel, though I think it may be more difficult than just "adding another mode" because the two VGA modes would be indistinguishable to Linux.

tf4ong, 2023-01-20:
Thanks, yea. NothingThanks much can be.

closed 2023-01-20 by tf4ong

RolandasRazma, 2023-01-22:
@tf4ong where do you go from here? I'm starting to think about forking driver

kbingham, 2023-01-23:
@RolandasRazma No need to 'fork' the driver - If you explore the new modes, and add it to the kernel, submit a PR to the raspberry pi kernel - or better yet - to the Linux Media subsystem to get the new mode added to the kernel itself.

davidplowman, 2023-01-23:
We'd certainly be happy to review and accept updates. There's the slight problem that as far as V4L2 is concerned, the existing 640x480 mode (up to 60fps) and a 90fps version would look the same, and there's no way, AIUI, to select one over the other. @kbingham Does that sound right, or do you think there's a workaround?

kbingham, 2023-01-23:
Sounds like a discussion to have in the Linux media mailing list, ideally with a list of the registers that need to be updated to facilitate the differences.

It should be supportable. The question is how much development is required to be able to do so

tf4ong, 2023-01-23:
@RolandasRazma  I don't have any plans to update the driver, I am not sure how to change the codes at that level