raspberrypi/picamera2 #526: [HOW-TO] Know USB camera resolution

RobertFlatt, 2023-01-17:
**Please only ask one question per issue!**
I want to get the best possible usb resolution, for still image capture. How do I know what I can specify?

**Describe what it is that you want to accomplish**
By default usb camera resolution appears to be 640x480 for both YUYV and MJPEG
This seems good for a `4:3` preview.
I want higher resolution for a still capture, I expect the resolution to depend on the usb camera used.

**Describe alternatives you've considered**
I tried setting a plausible, and higher resolution:
If in a still_configuration I set `main = {'size' : (1440, 1080)}` 
I get a capture that is `1600 x 896` , so it seems there is more resolution to be had.
But I asked for `4:3` and got `16:9`
Configuring higher resolutions, showed similar behavior. 

**Additional context**
Add any other context or screenshots about the request here.

RolandasRazma, 2023-01-17:
can you use `.sensor_modes` and sort them?

https://github.com/raspberrypi/picamera2/blob/af75efd3b76479a2f6401dda120a67f3ee417eea/examples/pick_mode.py

RobertFlatt, 2023-01-17:
For a usb cameras I tried `picam2.sensor_modes` only contains 'format' , tried two Logitech (very common) usb cameras, for each all that is returned by sensor_modes is two formats:
```
{'format': 'MJPEG'}
{'format': 'YUYV'}
```
Experiment shows that picamera2 generates a stream from a usb camera that can have discrete values of "size". 
How to enumerate these discrete values?
Is it possible to query libcamera directly?

RolandasRazma, 2023-01-18:
what about
`v4l2-ctl --list-formats-ext -d /dev/video0` ?

davidplowman, 2023-01-18:
Hi, I'm afraid USB camera support is limited with Picamera2, and we have no plans to extend it at the moment. You could try whether `picam2.sensor_resolution` will give you the maximum output resolution . You can of course use the regular V4L2 style interface, such as `v4l2-ctl`, too.

closed 2023-01-19 by RobertFlatt