raspberrypi/picamera2 #533: [HOW-TO] rotate image before saving

uselessgames, 2023-01-19:
**Describe what it is that you want to accomplish**
I am attempting to take a timelapse video with stamped time. I have successfully done this (code below). But the issue is my camera, because of poor design in the mount, is upside down. Easiest solution is to rotate programatically. If i rotate the video after the timelapse is taken the timestamp is now upside down. . haha 
the example doesnt seem to work on my sample code.  rotate https://github.com/raspberrypi/picamera2/blob/75385a298b7f7afc0e947cb0f321175cc27d8217/examples/rotation.py#L14

### how could i accomplish rotation of the image before saving please?

**Describe alternatives you've considered**
n/a

**Additional context**

1. i am using os lite headless (no monitor available). 
2. current script to take timelapse with stamped time. pardon my shitty code i am not a programmer;

```
#!/usr/bin/python3
import subprocess
import time
import os
import cv2

from picamera2 import Picamera2, MappedArray
from picamera2.encoders import MJPEGEncoder, Quality
from picamera2.outputs.fileoutput import FileOutput
from os.path import expanduser

# Set the parameters for the timelapse and timestamp
speedup_factor = 10
framerate = 3.0
resolution = (640,480)
colour = (0, 255, 0)
origin = (0, 30)
font = cv2.FONT_HERSHEY_SIMPLEX
scale = 1
thickness = 2

# set file and folders
#filename = time.strftime("%Y%m%d-%H%M%S")
filename = "speed" + str(speedup_factor) + "framerate" + str(framerate) #only for testing diff lapsed params
folder = expanduser("~pi") + "/recordings/"
file_mjpeg = folder + filename + ".mjpeg"
file_mp4 = folder + filename + ".mp4"

# Define an output which divides all the timestamps by a factor
class TimelapseOutput(FileOutput):
    def __init__(self, file=None, pts=None, speed=10):
        self.speed = int(speed)
        super().__init__(file, pts)

    def outputtimestamp(self, timestamp):
        if timestamp == 0:
            # Print timecode format for the first line
            print("# timestamp format v2", file=self.ptsoutput, flush=True)
        # Divide each timestamp by factor to speed up playback
        timestamp //= self.speed
        super().outputtimestamp(timestamp)

def apply_timestamp(request):
    timestamp = time.strftime("%Y-%m-%d %X")
    with MappedArray(request, "main") as m:
        cv2.putText(m.array, timestamp, origin, font, scale, colour, thickness)

picam2 = Picamera2()
config = picam2.create_video_configuration(main={"size": resolution})
picam2.configure(config)

picam2.pre_callback = apply_timestamp
encoder = MJPEGEncoder()
output = TimelapseOutput(file_mjpeg, folder + "timestamps.txt", speedup_factor)
encoder.output = output

picam2.start()

# Give time for Aec and Awb to settle, before disabling them
time.sleep(1)
picam2.set_controls({"AeEnable": False, "AwbEnable": False, "FrameRate": framerate})
# And wait for those settings to take effect
time.sleep(1)

picam2.start_encoder(encoder, quality=Quality.VERY_HIGH)

time.sleep(60)

picam2.stop_encoder()
picam2.stop()

# Create the output mp4 video **doesnt work!
#merge = subprocess.Popen([
#    "mkvmerge", "-o", "timelapse.mp4",
#    "--timestamps", "0:timestamps.txt",
#    "test.mjpeg"
#])
#merge.wait()

# seems easier to convert directly with ffmpeg... 
ffmpeg_convert =  "ffmpeg -i " + file_mjpeg + " -pix_fmt yuv420p -b:v 4000k -c:v libx265 -crf 28 -vf transpose=2,transpose=2 " + file_mp4
print(ffmpeg_convert)
#os.system(ffmpeg_convert)

time.sleep(3)
os.system("rm " + folder + "*.txt")
#os.system("rm " + folder + "*.mjpeg")

print("byeeee")

```

davidplowman, 2023-01-20:
Hi, if it's just a 180 degree rotation you're after, you can use:
```
from libcamera import Transform
```
and then
```
config = picam2.create_video_configuration(main={"size": resolution}, transform=Transform(180))
```
If that doesn't work, perhaps post a really minimal example that simply brings up a preview window and that shows any problems. Thanks!

uselessgames, 2023-01-20:
i am not able to use preview with headless.

regardless that does the trick. 

would be nice if the examples were consistent.

closed 2023-01-20 by uselessgames