raspberrypi/picamera2 #536: [HOW-TO] run the capture_timelapse example

uselessgames, 2023-01-23:
**Describe what it is that you want to accomplish**
posting as how to as unsure if it's a bug. when i try to run the [capture_timelapse](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_timelapse_video.py) example without any modifications it returns the error

`Error: The type of file 'test.mjpeg' could not be recognized.`

**Describe alternatives you've considered**
tried searching online but found nothing in relation to the picamera2 module. 

**Additional context**
rpi 4b, os lite, headless

full log as it appear on terminal. 
```
pi@raspberrypi:~ $ python test_convert.py
[0:35:39.113766724] [866]  INFO Camera camera_manager.cpp:299 libcamera v0.0.2+55-5df5b72c
[0:35:39.145105265] [867]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media4 and ISP device /dev/media0
[0:35:39.149574229] [866]  INFO Camera camera.cpp:1026 configuring streams: (0) 1920x1080-XBGR8888
[0:35:39.149901671] [867]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
mkvmerge v52.0.0 ('Secret For The Mad') 64-bit
Error: The type of file 'test.mjpeg' could not be recognized.

```

davidplowman, 2023-01-24:
Thanks for reporting this. Indeed I see the same thing. Actually I'm wondering about the choice of the MJPEGEncoder, it's not normally the codec you'd use with an mp4 or mkv file. I tried the H264Encoder instead and that seems to work fine.

@chrisruk Any thoughts on this, do you think we should be using h264 instead?

chrisruk, 2023-01-24:
@davidplowman I'm not sure I understand.  Doesn't capture_timelapse.py write a sequence of regular jpegs, at least that's what it seems to do for me?

I couldn't seem to find test_convert.py

davidplowman, 2023-01-24:
So the file in question is [this one](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_timelapse_video.py). Looking at it, I see it was added during last summer. I wonder if the author was inadvertently winding up with h.264 without realising it - there was a period of time when that was happening by accident when you asked for MJPEG.

So the answer here is probably to change this example to h.264 and go with that, or stick with MJPEG and not convert it. @uselessgames Does that work for you?

chrisruk, 2023-01-24:
Oh sorry! I was looking at a wrong example, which generated an mjpeg and I played that.

I get the same error too.

uselessgames, 2023-01-24:
turns out i had defined the encoder wrong `encoder = h264Encoder(10000000)` when it should have been `encoder = H264Encoder(10000000)`

uselessgames, 2023-01-24:
yes, also works. thank you

but now that i test i get better file compression with ffmpeg. `-vcodec libx265 -crf 28`

with H265Encoder file size 630KB
with ffmpeg file size 52 KB

when converting the same 20 second mjpeg test. 

since i'm doing 8 hour timelapse you can see how ffmpeg is preferred.

closed 2023-01-24 by uselessgames

davidplowman, 2023-01-24:
Can you say what you mean by "better compression"? Do you mean that the `H264Encoder` generates files that are too big, or the quality is not high enough. You can change the magic bitrate number (that 10000000 value) if you wish. Though a real-time H264 hardware encoder isn't going to do as well as offline encoders (also note there's no H265 encode available).

Anyway I'll push a PR to update this example, but then I think we're probably done? Thanks everyone!

uselessgames, 2023-01-24:
> Can you say what you mean by "better compression"? Do you mean that the H264Encoder generates files that are too big,

yes. when i convert the same mpjpeg using your example and ffmpeg the file size is significantly smaller in size with ffmpeg. 

see my previous comment. 

> for me now that i test i get better file compression with ffmpeg. `-vcodec libx265 -crf 28` 

> with H265Encoder file size 630KB
> with ffmpeg file size 52 KB

> when converting the same 20 second mjpeg test.

davidplowman, 2023-01-24:
https://github.com/raspberrypi/picamera2/pull/538