raspberrypi/picamera2 #540: [BUG]

tennisparty, 2023-01-25:
**Please only report one bug per issue!**

**Describe the bug**
I am running the lores stream for processing. I am running the main stream for a webserver and simultaneously recording via the MJPEG encoder (but not accessing the frames from the frameserver). I am collecting both the lores and main camera stream via the following method described in the docs `array = picam2.capture_arrays(["main", "lores"])` and sending these to different threads using the the frameserver.py example. The error is not consistent but seems to occur when a recording is triggered. I am not using a preview.

**To Reproduce**
See above.

**Expected behaviour**
I didn't expect this error message

**Console Output, Screenshots**
```callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 74, in handle_request
    completed_request = picam2.process_requests()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1023, in process_requests
    self._encoder.encode(stream, display_request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 243, in encode
    fcntl.ioctl(self.vd, VIDIOC_QBUF, buf)```

**Hardware :**
pi 3a+ with ov9281

**Additional context**
Add any other context about the problem here.

davidplowman, 2023-01-26:
Hi, thanks for the report. It sounds as though the encoder has perhaps not started, but it's very hard to diagnose in the abstract. Are you able to attach a minimal test case that shows the problem? Thanks.

tennisparty, 2023-01-26:
Hi David,

Here is an example code which still raises the error

```
#!/usr/bin/python3

# These two are only needed for the demo code below the FrameServer class.
import time
from threading import Condition, Thread

from picamera2 import Picamera2
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import CircularOutput, FileOutput

class FrameServer:
    def __init__(self, picam2, stream='main'):
        """A simple class that can serve up frames from one of the Picamera2's configured streams to multiple other threads.
        Pass in the Picamera2 object and the name of the stream for which you want
        to serve up frames.
        """
        self._picam2 = picam2
        self._stream = stream
        self._array = None
        self._condition = Condition()
        self._running = True
        self._count = 0
        self._thread = Thread(target=self._thread_func, daemon=True)

    @property
    def count(self):
        """A count of the number of frames received."""
        return self._count

    def start(self):
        """To start the FrameServer, you will also need to start the Picamera2 object."""
        self._thread.start()

    def stop(self):
        """To stop the FrameServer
        First stop any client threads (that might be
        blocked in wait_for_frame), then call this stop method. Don't stop the
        Picamera2 object until the FrameServer has been stopped.
        """
        self._running = False
        self._thread.join()

    def _thread_func(self):
        while self._running:
            array = picam2.capture_arrays(["main", "lores"])
            self._count += 1
            with self._condition:
                self._array = array
                self._condition.notify_all()

    def wait_for_frame(self, previous=None):
        """You may optionally pass in the previous frame that you got last time you called this function.
        This will guarantee that you don't get duplicate frames
        returned in the event of spurious wake-ups, and it may even return more
        quickly in the case where a new frame has already arrived.
        """
        with self._condition:
            if previous is not None and self._array is not previous:
                return self._array
            while True:
                self._condition.wait()
                if self._array is not previous:
                    return self._array

# Below here is just demo code that uses the class:

def thread1_func():
    global thread1_count
    frame = None
    while not thread_abort:
        (frame, lores), metadata  = server.wait_for_frame(frame)
        thread1_count += 1

def thread2_func():
    global thread2_count
    frame = None
    while not thread_abort:
        (frame, lores), metadata  = server.wait_for_frame(frame)
        thread2_count += 1
        if not thread2_count % 50000:
            print("thread2_count", thread2_count)
            print("entering recording loop")
            recording()

def recording():
    global encoding
    global ltime
    if not encoding:
        ltime = time.time()
        encoder.output = FileOutput("{}.mjpeg".format(time.strftime("%Y%m%d-%H%M%S")))
        picam2.start_encoder()
        print("recording for 2.0 seconds after last motion event")
        encoding = True
    else:
        if encoding and time.time() - ltime > 2.0:
            print("encoding is less that time")
            picam2.stop_encoder()
            print("recording stopped")
            encoding = False
            print("ending recording loop")

thread_abort = False
thread1_count = 0
thread2_count = 0
thread1 = Thread(target=thread1_func)
thread2 = Thread(target=thread2_func)

picam2 = Picamera2()
lores = (640, 400)
config = picam2.create_video_configuration(controls={"FrameDurationLimits": (33333, 33333)},main={'format': 'YUV420',"size": (1280, 720)}, lores={"size": lores})
picam2.align_configuration(config)
picam2.configure(config)
encoder = MJPEGEncoder()
picam2.encoder = encoder
picam2.start()

encoding = False
ltime = 0

server = FrameServer(picam2)
server.start()

thread1.start()
thread2.start()

time.sleep(25)

thread_abort = True
thread1.join()
thread2.join()
server.stop()
picam2.stop()

print("Thread1 received", thread1_count, "frames")
print("Thread2 received", thread2_count, "frames")
print("Server received", server.count, "frames")
```

error output
```
ending recording loop
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 74, in handle_request
    completed_request = picam2.process_requests()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1023, in process_requests
    self._encoder.encode(stream, display_request)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/v4l2_encoder.py", line 243, in encode
    fcntl.ioctl(self.vd, VIDIOC_QBUF, buf)
OSError: [Errno 22] Invalid argument
thread2_count 150000
entering recording loop
recording for 2.0 seconds after last motion event
thread2_count 200000
entering recording loop
...
```

davidplowman, 2023-01-26:
If anything, it looks like maybe it's stopped the encoder, but then tried getting it to encode another frame. I know there was a problem in that area some time ago, can I just check that you've got the most up-to-date version of Picamera2? Thanks.

davidplowman, 2023-01-26:
The other little thing that I noticed is that your threads are all busy-spinning. The problem is that the test
```
            if previous is not None and self._array is not previous:
                return self._array
```
in `wait_for_frame` no longer works correctly now that you're returning a list of things rather than a single frame. You could probably just delete those lines and the behaviour will be more normal, though I'm guessing that
```
        if not thread2_count % 50000:
```
is no longer an appropriate test. It might be worth fixing that up and re-posting if you can. Thanks!

tennisparty, 2023-01-27:
Hi David, the update seems to have solved the issue, it is not recurring now, in case it is useful picamera2 was on 0.3.5 and is now on 0.3.8.

For the sake of completeness I have tried modifying the code to check that the output from the `array = picam2.capture_arrays(["main", "lores"])` statement is the same in the wait_for_frame function

```
#!/usr/bin/python3

# These two are only needed for the demo code below the FrameServer class.
import time
from threading import Condition, Thread

import numpy as np

from picamera2 import Picamera2
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import CircularOutput, FileOutput

class FrameServer:
    def __init__(self, picam2, stream='main'):
        """A simple class that can serve up frames from one of the Picamera2's configured streams to multiple other threads.
        Pass in the Picamera2 object and the name of the stream for which you want
        to serve up frames.
        """
        self._picam2 = picam2
        self._stream = stream
        self._array = None
        self._condition = Condition()
        self._running = True
        self._count = 0
        self._thread = Thread(target=self._thread_func, daemon=True)

    @property
    def count(self):
        """A count of the number of frames received."""
        return self._count

    def start(self):
        """To start the FrameServer, you will also need to start the Picamera2 object."""
        self._thread.start()

    def stop(self):
        """To stop the FrameServer
        First stop any client threads (that might be
        blocked in wait_for_frame), then call this stop method. Don't stop the
        Picamera2 object until the FrameServer has been stopped.
        """
        self._running = False
        self._thread.join()

    def _thread_func(self):
        while self._running:
            array = picam2.capture_arrays(["main", "lores"])
            # array = self._picam2.capture_array(self._stream)
            self._count += 1
            with self._condition:
                self._array = array
                self._condition.notify_all()

    def wait_for_frame(self, previous=None):
        """You may optionally pass in the previous frame that you got last time you called this function.
        This will guarantee that you don't get duplicate frames
        returned in the event of spurious wake-ups, and it may even return more
        quickly in the case where a new frame has already arrived.
        """
        with self._condition:
            if previous is None:
                return self._array
            if previous is not None and np.array_equal(self._array[0], previous[0]) or not np.array_equal(self._array[1], previous[1]):
                return self._array
            while True:
                self._condition.wait()
                if not np.array_equal(self._array[0], previous[0]) or not np.array_equal(self._array[1], previous[1]):
                    return self._array

# Below here is just demo code that uses the class:

def thread1_func():
    global thread1_count
    output = None
    while not thread_abort:
        output  = server.wait_for_frame(output)
        # print(output)
        if output is not None:
            (main, lores), metadata  = output
            # print(main)
            # print(type(main))
            # print(main.shape)
            # print(lores)
            # print(type(lores))
            # print(lores.shape)
        thread1_count += 1
        print("t1", thread1_count)

def thread2_func():
    global thread2_count
    output = None
    while not thread_abort:
        output  = server.wait_for_frame(output)
        # print(output)
        if output is not None:
            (main, lores), metadata  = output
            # print(main)
            # print(type(main))
            # print(main.shape)
            # print(lores)
            # print(type(lores))
            # print(lores.shape)
        thread2_count += 1
        print("t2", thread2_count)
        if not thread2_count % 100:
            print("thread2_count", thread2_count)
            print("entering recording loop")
            recording()

def recording():
    global encoding
    global ltime
    if not encoding:
        ltime = time.time()
        encoder.output = FileOutput("{}.mjpeg".format(time.strftime("%Y%m%d-%H%M%S")))
        picam2.start_encoder()
        print("recording for 2.0 seconds after last motion event")
        encoding = True
    else:
        if encoding and time.time() - ltime > 2.0:
            print("encoding is less that time")
            picam2.stop_encoder()
            print("recording stopped")
            encoding = False
            print("ending recording loop")

thread_abort = False
thread1_count = 0
thread2_count = 0
thread1 = Thread(target=thread1_func)
thread2 = Thread(target=thread2_func)

picam2 = Picamera2()
lores = (640, 400)
config = picam2.create_video_configuration(controls={"FrameDurationLimits": (33333, 33333)},main={'format': 'YUV420',"size": (1280, 720)}, lores={"size": lores})
picam2.align_configuration(config)
picam2.configure(config)
encoder = MJPEGEncoder()
picam2.encoder = encoder
picam2.start()

encoding = False
ltime = 0

server = FrameServer(picam2)
server.start()

thread1.start()
thread2.start()

time.sleep(25)

thread_abort = True
thread1.join()
thread2.join()
server.stop()
picam2.stop()

print("Thread1 received", thread1_count, "frames")
print("Thread2 received", thread2_count, "frames")
print("Server received", server.count, "frames")
```

This seems to be working but my threads are receiving quite different numbers
```
Thread1 received 1600 frames
Thread2 received 1251 frames
Server received 745 frames
```

would you expect this?

davidplowman, 2023-01-27:
Thanks for the updated code. Glad that the later Picamera2 version seems to be better!

Nonetheless, I still can't see how it could be correct for either thread to report more frames than the server itself, so I think something must still be wrong. I must confess that this snippet from `wait_for_frame`:
```
            if previous is not None and np.array_equal(self._array[0], previous[0]) or not np.array_equal(self._array[1], previous[1]):
                return self._array
```
looks slightly scary. `np.array_equal` could be going off and doing per-pixel comparisons, whereas what you really want is "was this the same thing I handed out last time" (in C/C++ it might just be a simple pointer test). I think in in Python you might find that this
```
            if previous is not None and self._array is not previous:
                return self._array
```
would work. Maybe give that a try and see if the frame counts look sensible?

tennisparty, 2023-01-27:
Thanks for looking, when I remove the
```
        if not thread2_count % 100:
            print("thread2_count", thread2_count)
            print("entering recording loop")
            recording()
```
call from thread2_func I receive
```
Thread1 received 1595 frames
Thread2 received 1505 frames
Server received 745 frames
```

If I still call the record loop and edit the wait_for_frame function in the class so that it checks for `if previous is not None and self._array is not previous:` rather than numpy arrays, I get the result:
```
Thread1 received 2447 frames
Thread2 received 1828 frames
Server received 746 frames
```
If I remove the record loop I get
```
Thread1 received 1690 frames
Thread2 received 1589 frames
Server received 745 frames
```
So it seems to be calling that calling the record function in thread2 is altering the number of frames received

I've tested a version of the original frameserver code with the record loop added:
```
#!/usr/bin/python3

# These two are only needed for the demo code below the FrameServer class.
import time
from threading import Condition, Thread

from picamera2 import Picamera2
from picamera2.encoders import MJPEGEncoder
from picamera2.outputs import CircularOutput, FileOutput

class FrameServer:
    def __init__(self, picam2, stream='main'):
        """A simple class that can serve up frames from one of the Picamera2's configured streams to multiple other threads.
        Pass in the Picamera2 object and the name of the stream for which you want
        to serve up frames.
        """
        self._picam2 = picam2
        self._stream = stream
        self._array = None
        self._condition = Condition()
        self._running = True
        self._count = 0
        self._thread = Thread(target=self._thread_func, daemon=True)

    @property
    def count(self):
        """A count of the number of frames received."""
        return self._count

    def start(self):
        """To start the FrameServer, you will also need to start the Picamera2 object."""
        self._thread.start()

    def stop(self):
        """To stop the FrameServer
        First stop any client threads (that might be
        blocked in wait_for_frame), then call this stop method. Don't stop the
        Picamera2 object until the FrameServer has been stopped.
        """
        self._running = False
        self._thread.join()

    def _thread_func(self):
        while self._running:
            array = self._picam2.capture_array(self._stream)
            self._count += 1
            with self._condition:
                self._array = array
                self._condition.notify_all()

    def wait_for_frame(self, previous=None):
        """You may optionally pass in the previous frame that you got last time you called this function.
        This will guarantee that you don't get duplicate frames
        returned in the event of spurious wake-ups, and it may even return more
        quickly in the case where a new frame has already arrived.
        """
        with self._condition:
            if previous is not None and self._array is not previous:
                return self._array
            while True:
                self._condition.wait()
                if self._array is not previous:
                    return self._array

# Below here is just demo code that uses the class:

def thread1_func():
    global thread1_count
    while not thread_abort:
        _ = server.wait_for_frame()
        thread1_count += 1

def thread2_func():
    global thread2_count
    frame = None
    while not thread_abort:
        frame = server.wait_for_frame(frame)
        thread2_count += 1
        if not thread2_count % 100:
            print("thread2_count", thread2_count)
            print("entering recording loop")
            recording()

def recording():
    global encoding
    global ltime
    if not encoding:
        ltime = time.time()
        encoder.output = FileOutput("{}.mjpeg".format(time.strftime("%Y%m%d-%H%M%S")))
        picam2.start_encoder()
        print("recording for 2.0 seconds after last motion event")
        encoding = True
    else:
        if encoding and time.time() - ltime > 2.0:
            print("encoding is less that time")
            picam2.stop_encoder()
            print("recording stopped")
            encoding = False
            print("ending recording loop")

thread_abort = False
thread1_count = 0
thread2_count = 0
thread1 = Thread(target=thread1_func)
thread2 = Thread(target=thread2_func)

picam2 = Picamera2()
lores = (640, 400)
config = picam2.create_video_configuration(controls={"FrameDurationLimits": (33333, 33333)},main={'format': 'YUV420',"size": (1280, 720)}, lores={"size": lores})
picam2.align_configuration(config)
picam2.configure(config)
encoder = MJPEGEncoder()
picam2.encoder = encoder
picam2.start()

encoding = False
ltime = 0

server = FrameServer(picam2)
thread1.start()
thread2.start()
server.start()
# picam2.sta

time.sleep(25)

thread_abort = True
thread1.join()
thread2.join()
server.stop()
picam2.stop()

print("Thread1 received", thread1_count, "frames")
print("Thread2 received", thread2_count, "frames")
print("Server received", server.count, "frames")
```
and here calling the record loop doesn't affect the number of frames received, so something must be up!
```
Thread1 received 745 frames
Thread2 received 745 frames
Server received 746 frames
```

tennisparty, 2023-01-27:
scrap, that I think I have found the culprit, I had added a return check if previous was None in the wait_for_frame function. The first time the record loop was called previous was none and this altered the number of frames received for this call. It works OK if altered as follows:
```
    def wait_for_frame(self, previous=None):
        """You may optionally pass in the previous frame that you got last time you called this function.
        This will guarantee that you don't get duplicate frames
        returned in the event of spurious wake-ups, and it may even return more
        quickly in the case where a new frame has already arrived.
        """
        with self._condition:
            if previous is not None and self._array is not previous:
                return self._array
            while True:
                self._condition.wait()
                if self._array is not previous:
                    return self._array
```

davidplowman, 2023-01-27:
Cool, are you getting basically the same number of frames reported everywhere now? That sounds much better!

tennisparty, 2023-01-27:
Yes, all the numbers are matching and they all change together if I alter the frame rate, thanks again for the help

davidplowman, 2023-02-03:
I think I can probably close this one now. Of course, if anything else comes up, just open a new issue!

closed 2023-02-03 by davidplowman