raspberrypi/picamera2 #542: [HOW-TO] Blitting video to PYGAME while recording

eat-sleep-code, 2023-01-26:
What is the most efficient way to switch between preview and VIDEO capture and output both to PYGAME?

For example, if in a PYGAME app, I want to show preview.   Then click an on-screen button, and recording would start, still showing the view of the video on the screen, until the button is clicked again.  At the point, the video recording would stop and return to the preview.

Below you will see in `captureImage`, I am pulling the data from the request and blitting the single frame to the screen.   Is the same thing possible in video mode?   
```
# ------------------------------------------------------------------------------

def captureImage(filepath, raw = True):
	request = camera.switch_mode_and_capture_request('still')
	request.save('main', filepath)
	if raw == True:
		filepathDNG = filepath.replace('.jpg', '.dng')
		request.save_dng(filepathDNG)
	array = request.make_array('main')
	request.release()
	capturedFrame = pygame.image.frombuffer(array.data, camera.sensor_resolution, 'RGB')
	globals.displaySurface.blit(capturedFrame, (0, 0))
	pygame.display.update()
	
	time.sleep(5)
	return

# ------------------------------------------------------------------------------

def captureVideo(filepath):
	global isRecording
	
	encoder = H264Encoder(bitrate=1000000)
	camera.start_recording(encoder, filepath, config="video")
	if isRecording == False:
		camera.stop_recording()
	return

# ------------------------------------------------------------------------------
```

davidplowman, 2023-01-27:
Hi, I'm not totally sure if I understand what you want to do, but while a video is recording you can continue to capture requests and save them or display them just as you did before. What you can't do, though, is switch to a different camera mode as that will kill the recording (in fact, something bad might even happen, so I need to try that). So `capture_request` would be OK, `switch_mode_and_capture_request` would not be.

There's an example [here](https://github.com/raspberrypi/picamera2/blob/main/examples/still_during_video.py) that records a video and captures a still while the recording is running. Have I answered the right question?

eat-sleep-code, 2023-01-28:
So I can keep blitting my preview image at the same time as I am recording?    Do the two need to have matching resolutions?

davidplowman, 2023-01-28:
Yes you can. The camera system will give you up to 2 different output stream, the "main" and "lores" stream. The "main" stream can be a YUV or RGB format, and the "lores" one must be a YUV format of the same or lower resolution than the "main" stream.

The "lores" stream can be less convenient for showing as a preview unless you're using the QTGL or DRM previews which do support YUV420. But most other Python modules don't "like" YUV very much. You can convert it to RGB using software (OpenCV has a function to do this), but obviously it eats your CPU.

The example I linked is recording the "lores" stream and then captures a higher resolution still image from the "main" stream.

davidplowman, 2023-02-13:
Hi, just wondering if there's anything else to investigate on this one? Thanks!

eat-sleep-code, 2023-02-13:
I have been wrapped up in a bunch of other items and have not been able to circle back to this yet.

davidplowman, 2023-04-04:
Hi again, just wondering if we should close this one for now? It's easy enough to open a fresh issue if something else comes up. Thanks!

eat-sleep-code, 2023-04-09:
Go ahead and close for now

davidplowman, 2023-04-11:
Thanks, please do open a new ticket if anything else comes up.