raspberrypi/picamera2 #545: [HOW-TO] Is there a way to add / update exif metadata

mattjlewis, 2023-01-29:
I would like to augment the JPEG image file with additional metadata, specifically GPS location and the time that the photo was taken. I can add these via e.g. exiv2 (`Exif.Image.DateTime`, `Exif.Image.DateTimeOriginal`, `Exif.GPSInfo.GPSLatitudeRef`, `Exif.GPSInfo.GPSLatitude`, `Exif.GPSInfo.GPSLongitudeRef`, `Exif.GPSInfo.GPSLongitude`), however, I cannot see a way to add directly.

I have tried altering the metadata in a pre_callback method but without luck, e.g. :

```python
def add_metadata(request):
    metadata = request.get_metadata()
    print("In pre_callback ExposureTime: " + str(metadata["ExposureTime"]))
    metadata["DateTime"] = "2023:01:29 16:25:00"
    metadata["DateTimeOriginal"] = "2023:01:29 16:25:00"
    print("In pre_callback DateTime: " + str(metadata["DateTime"]))
    print("In pre_callback DataTimeOriginal: " + str(metadata["DateTimeOriginal"]))
```

davidplowman, 2023-02-01:
Hi, thanks for the question. Indeed, I don't think there's a built-in way to accomplish this at the moment, and as you've discovered, you can't add arbitrary values to the image metadata and have the code understand what to do with them.

We can look into the question, but in the meantime is there a workaround where you write the JPEG to a memory buffer, then use one of Python's EXIF packages to amend it, and then re-save it to file?

mattjlewis, 2023-02-02:
Hi, thanks for the quick reply. Yes - currently doing a call out to exiv2. Would be really nice if at least DateTime / DateTimeOriginal was written by picamera2.

davidplowman, 2023-02-03:
Yes, I'll certainly add the dates. For adding custom info, that probably just needs a bit of plumbing. We'll put it on the list!

mattjlewis, 2023-02-03:
Brilliant, thank you.

davidplowman, 2023-02-03:
https://github.com/raspberrypi/picamera2/pull/552

richardthegit, 2023-09-20:
I'd like to second this. The problem with patching the EXIF after file generation is that you tend to end up compressing the file twice.

pinheiromelobruno, 2025-02-27:
I would like to know if this issue has evolved. I particularly need the GPS metadata. A bit of context, I intend to employ the AI ​​cam to make detections and record the geographic coordinates from where they were obtained.

davidplowman, 2025-02-27:
Some of the methods like `picam2.capture_file(...)` have an `exif_data` argument that looks like it should accept a dictionary using piexif fields. Is that what you're looking for?