raspberrypi/picamera2 #548: [BUG] Cannot set lores through configuration any longer after last update

JooJooBee666, 2023-01-30:
**Please only report one bug per issue!**

**Describe the bug**
Since updating to the latest python3-libcamera (0~git20230124+9b860a66-1), I can no longer initialize the camera with a config that specifies the lores size.

**To Reproduce**
```
from picamera2 import Picamera2
picam2 = Picamera2()

pc1 = picam2.create_video_configuration(main={'size': (1664, 1248)},controls={'FrameDurationLimits': (142857, 142857)},lores={'size': (640, 480)}, buffer_count=10)
pc2 = picam2.create_video_configuration(main={'size': (1664, 1248)},controls={'FrameDurationLimits': (142857, 142857)}, buffer_count=10)
```
Take a look at the difference between pc1 and pc2.  The only think different is the missing "lores=" in the second.  Running picam2.config(pc1) generates the following error:
```
>>> picam2.config(pc1)
[4850] ERROR Camera camera.cpp:1014 Can't configure camera with invalid configuration
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 976, in configure
    self.configure_(camera_config)
  File "/usr/local/lib/python3.9/dist-packages/picamera2/picamera2.py", line 913, in configure_
    raise RuntimeError("Configuration failed: {}".format(camera_config))
RuntimeError: Configuration failed: {'use_case': 'video', 'transform': <libcamera.Transform 'identity'>, 'colour_space': <libcamera.ColorSpace 'Rec709/Rec709/None/Full'>, 'buffer_count': 10, 'queue': True, 'main': {'format': 'XBGR8888', 'size': (1664, 1248), 'stride': 6656, 'framesize': 8306688}, 'lores': {'format': 'YUV420', 'size': (640, 480), 'stride': 640, 'framesize': 460800}, 'raw': None, 'controls': {'NoiseReductionMode': <NoiseReductionModeEnum.Fast: 1>, 'FrameDurationLimits': (142857, 142857)}, 'display': 'main', 'encode': 'main'}
```

However, omitting the lores= and using the pc2 configuration is fine:
```
>>> picam2.configure(pc2)
[1:22:26.704975834] [4850]  INFO Camera camera.cpp:1026 configuring streams: (0) 1664x1248-XBGR8888
[1:22:26.706020105] [4854]  INFO RPI raspberrypi.cpp:767 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
```
This also fails when setting using **create_still_configuration** and **create_preview_configuration**.  Even a simple config such as **create_preview_configuration(lores={})** will fail to initialize.

If I initialize the video_configuration without the lores, then run **picam2.video_configuration.enable_lores()**, I can then set the lores size by doing **picam2.video_configuration.lores.size=**

**Expected behaviour**
I expect this to work as documented [here] (https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf) .  An application I am using relies on the ability to change this resolution. 

**Console Output, Screenshots**
See above.

**Hardware :**
Pi4 B 8GB

**Additional context**
As I stated, this was fine up until I installed the latest updates.  Reboots don't fix.

JooJooBee666, 2023-01-30:
I should note that after more playing around, I was able to get my things back up and working...I had to rollback the python3-libcamera AND libcamera0 packages both to the previous version I pulled from apt cache:

```
libcamera0:arm64 (0~git20221130+0684c373-1)
lpython3-libcamera (0~git20221130+0684c373-1)
```

njhollinghurst, 2023-02-01:
Are you using standard Raspberry Pi OS 64-bit?

There might be a mismatch between latest python3-libcamera and libcamera0. Investigating.

JooJooBee666, 2023-02-01:
Sorry I forgot to mention that, but yes, using the standard Bullseye install.  The python3-libcamera update changes the libcamera so to libcamera.so.0.0.3 (previously libcamera.so.0.0.2).  I'm assuming the python3-picamer2 install is aware of this as when I rolled back that package first, it complained about the missing libcamera.so.0.0.2 (during __init__).  So that's what made me roll that back too.  I suspect something changed in the libcamera3 in regards to the lores handling but no one caught it.  So that's why I filed this here.  I could be wrong though as I tried briefly to debug the is code, but haven't been able to trace the root of the issue just due to my lack of knowledge on libcamera.

njhollinghurst, 2023-02-01:
A clean 64-bit install, update, upgrade gives these versions:
```
libcamera-apps/stable,now 0~git20230125+9f08463-1 arm64 [installed]
libcamera-tools/stable,now 0~git20230124+9b860a66-1 arm64 [installed]
libcamera0/stable,now 0~git20230124+9b860a66-1 arm64 [installed,automatic]
python3-libcamera/stable,now 0~git20230124+9b860a66-1 arm64 [installed,automatic]
python3-picamera2/stable,stable,now 0.3.8-1 all [installed]
```
Can you try updating and upgrading all of those?

Of your examples, the last one - setting lores after configure - did not work for me. It needs another `picam2.configure("video")` to switch to the modified configuration.

JooJooBee666, 2023-02-01:
That's the combination of packages that breaks things (last update of them), which is what I'm referring to here with this bug.  I could try it again but I'm sure it will fail.

njhollinghurst, 2023-02-01:
Yes, I am able to capture lores images with that change, and also with your first `pc1`. On a Pi4 2GB with HQ Cam.

What is the result of `find /usr -name '*libcamera*'?  Are there any obvious duplicates (that are not symlinks) or anything in `/usr/local`?  Perhaps there are remnants of an older version somewhere.

JooJooBee666, 2023-02-01:
Well its not the latest (as we know), but here's what it looks like working right now:
```
/usr/bin/libcamera-jpeg
/usr/bin/libcamera-vid
/usr/bin/libcamera-still
/usr/bin/libcamera-raw
/usr/bin/libcamera-hello
/usr/lib/python3/dist-packages/libcamera
/usr/lib/python3/dist-packages/libcamera/_libcamera.cpython-39-aarch64-linux-gnu.so
/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
/usr/lib/aarch64-linux-gnu/libcamera_app.so
/usr/lib/aarch64-linux-gnu/libcamera
/usr/lib/aarch64-linux-gnu/libcamera-base.so.0.0.2
/usr/lib/aarch64-linux-gnu/libcamera_app.so.1.1.0
/usr/lib/aarch64-linux-gnu/libcamera.so.0.0.2
/usr/lib/aarch64-linux-gnu/libcamera_app.so.1
/usr/share/libcamera
/usr/share/doc/libcamera-tools
/usr/share/doc/libcamera-0.0.2
/usr/share/doc/libcamera0
/usr/share/doc/python3-libcamera
/usr/share/doc/libcamera-apps
/usr/local/lib/python3.11/site-packages/libcamera
/usr/local/lib/python3.11/site-packages/libcamera/_libcamera.so
/usr/local/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
/usr/local/lib/aarch64-linux-gnu/pkgconfig/libcamera-base.pc
/usr/local/lib/aarch64-linux-gnu/pkgconfig/libcamera.pc
/usr/bin/libcamera-jpeg
/usr/bin/libcamera-vid
/usr/bin/libcamera-still
/usr/bin/libcamera-raw
/usr/bin/libcamera-hello
/usr/lib/python3/dist-packages/libcamera
/usr/lib/python3/dist-packages/libcamera/_libcamera.cpython-39-aarch64-linux-gnu.so
/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
/usr/lib/aarch64-linux-gnu/libcamera_app.so
/usr/lib/aarch64-linux-gnu/libcamera
/usr/lib/aarch64-linux-gnu/libcamera-base.so.0.0.2
/usr/lib/aarch64-linux-gnu/libcamera_app.so.1.1.0
/usr/lib/aarch64-linux-gnu/libcamera.so.0.0.2
/usr/lib/aarch64-linux-gnu/libcamera_app.so.1
/usr/share/libcamera
/usr/share/doc/libcamera-tools
/usr/share/doc/libcamera-0.0.2
/usr/share/doc/libcamera0
/usr/share/doc/python3-libcamera
/usr/share/doc/libcamera-apps
/usr/local/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
/usr/local/lib/aarch64-linux-gnu/pkgconfig/libcamera-base.pc
/usr/local/lib/aarch64-linux-gnu/pkgconfig/libcamera.pc
```

njhollinghurst, 2023-02-02:
In your last code block there is the line
```
[2:04:51.247209241] [6700]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+24-d81505b8-dirty (2023-01-19T15:38:23-05:00)
```
With the latest version, I would expect to see `v0.0.3+40-9b860a66`. 

I believe this problem is fixed on the latest versions from `apt` (but perhaps not those from `pip`).

JooJooBee666, 2023-02-02:
Ok, I will remove the hold on the packages and update again today.  I'll report back my results once I do.

JooJooBee666, 2023-02-02:
Ok, unheld and updates/upgraded all packages.  Drumroll please... :drum: :drum: :drum: 
```
>>> from picamera2 import Picamera2
>>> picam2 = Picamera2()
[68:08:28.258800184] [57742]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[68:08:28.331874806] [68030]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
>>> pc1 = picam2.create_video_configuration(main={'size': (1664,1248)},controls={'FrameDurationLimits':(142857, 142857)},lores={'size': (640,480)})
>>> picam2.configure(pc1)
[68:13:26.244568473] [57742]  INFO Camera camera.cpp:1028 configuring streams: (0) 1664x1248-XBGR8888 (1) 640x480-YUV420
[68:13:26.245754850] [68030]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
>>> 

```
We have a winner! :trophy: 

@njhollinghurst Yep, looks like it's fixed now. Thanks for the assist!  :beers:

closed 2023-02-02 by JooJooBee666