raspberrypi/picamera2 #549: [HOW-TO] Multiple outputs with main and lores

azsde, 2023-02-01:
Hello,

I am trying to understand how the main and lores configuration would work with the mutliple output example:

```
from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput, FfmpegOutput
import time

picam2 = Picamera2()
video_config = picam2.create_video_configuration()
picam2.configure(video_config)

encoder = H264Encoder(repeat=True, iperiod=15)
output1 = FfmpegOutput("-f mpegts udp://<ip-address>:12345")
output2 = FileOutput()
encoder.output = [output1, output2]

# Start streaming to the network.
picam2.start_encoder(encoder)
picam2.start()
time.sleep(5)

# Start recording to a file.
output2.fileoutput = "test.h264"
output2.start()
time.sleep(5)
output2.stop()

# The file is closed, but carry on streaming to the network.
time.sleep(9999999)
``` 

What I would like to do, is to stream a "low resolution" video feed and record a "high resolution" video file, but I don't know if this is possible ?

davidplowman, 2023-02-02:
Currently Picamera2 supports encoding only one of the streams. You can of course encode the other too, but you'd have to do it "by hand". I can't think there's anything specifically stopping you from doing this, but have never tried it so that's just a bit of a guess!

azsde, 2023-02-02:
Alright, it would be a nice addition to picamera2 :)

davidplowman, 2023-02-03:
I've added a PR [here](https://github.com/raspberrypi/picamera2/pull/551) which shows how to do this (I found I wanted a couple of API tweaks to make it look "nicer").

So I'll probably close this one for now, but I agree that having Picamera2 itself handle both encodes might be interesting, so we'll put that on our list.

@chrisruk What do you think about have a list of encoders (or something) instead of a single one?

closed 2023-02-03 by davidplowman