raspberrypi/picamera2 #554: [BUG] Switching modes after recording causes hang

jonasoh, 2023-02-04:
I am experiencing a hang when switching between video and still image modes. This is on 64-bit bullseye, Raspberry Pi 3B+, Camera module v3.

**To Reproduce**
The following program causes a hang:
```
import time
from picamera2 import Picamera2
from picamera2.outputs import FileOutput
from picamera2.encoders import MJPEGEncoder

cam = Picamera2()
still_config = cam.create_still_configuration()
video_config = cam.create_video_configuration()

cam.configure(video_config)
cam.start_recording(MJPEGEncoder(), FileOutput('out.mjpeg'))
time.sleep(5)

cam.stop_recording()
cam.switch_mode(still_config)
cam.capture_file('test.png')
```

When breaking out with Ctrl+C, I get the following output:
```
Traceback (most recent call last):
  File "/home/pi/test.py", line 15, in <module>
    cam.switch_mode(still_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1215, in switch_mode
    return self.dispatch_functions(functions, wait, signal_function)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1163, in dispatch_functions
    return job.get_result() if wait else job
  File "/usr/lib/python3/dist-packages/picamera2/job.py", line 77, in get_result
    return self._future.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 435, in result
    self._condition.wait(timeout)
  File "/usr/lib/python3.9/threading.py", line 312, in wait
    waiter.acquire()
KeyboardInterrupt
```

This seems to be similar to the bug reported in #470 which was not resolved.

Here is the CMA usage, before running the script:
```
$ grep Cma /proc/meminfo
CmaTotal:         262144 kB
CmaFree:          252780 kB
```

During recording:
```
CmaTotal:         262144 kB
CmaFree:          217936 kB
```

And when it hangs at the `switch_mode` line:
```
CmaTotal:         262144 kB
CmaFree:          220676 kB
```

Any ideas?

chrisruk, 2023-02-06:
Thanks for the report, I can also replicate this on the 32-bit version of bullseye

davidplowman, 2023-02-13:
I think that `stop_recording` stops the camera, so the `switch_mode` after that will hang. The easiest fix might just be to call `stop_encoder` instead of `stop_recording`, but I agree that we could perhaps produce more helpful error messages in these circumstances. @chrisruk What do you think?

jonasoh, 2023-02-13:
I do not have access to the Pi this week, but in #470 it is mentioned that `switch_mode_and_capture_file` also hangs the camera. I am almost certain that I experienced this on my setup as well, using one of the `switch_mode_and_capture_*` functions.

davidplowman, 2023-02-13:
`switch_mode_and_capture_file` will have exactly the same problem as `switch_mode`. It requires the camera still to be running. Please try `stop_encoder` instead of `stop_recording` and let us know if that works better. Thanks!

jonasoh, 2023-02-13:
Yes, it might be a separate issue from the previous report, which found that a hang would be triggered by `switch_mode_and_capture_file` without a prior call to `stop_recording` on a 64 MP camera (https://github.com/raspberrypi/picamera2/issues/470#issuecomment-1342741542). I'll report back when I have access to the machine again.

jonasoh, 2023-02-21:
Hi, just got back to test this out and it indeed works as expected without the call to `stop_recording`. Thanks @davidplowman!

davidplowman, 2023-02-22:
Thanks for the update!

closed 2023-02-22 by davidplowman