raspberrypi/picamera2 #555: [BUG]

johncblacker, 2023-02-05:
**Please only report one bug per issue!**

**Describe the bug**
The example for setting frame rate at the end of section 4.3 of the doc, page 22 doesn't work.  Response is: "AttributeError: 'dict' object has not attribute 'FrameRate'

**To Reproduce**
>>>from picamera2 import Picamera2
>>>picam2 = Picamera2()
>>>picam2.video_configuration.controls.FrameRate = 30
Traceback (most recent call last):
  File "<stdin>", Line 1, in <module>
AttributeError: 'dict' object has no attribute 'FrameRate'
>>>

**Expected behaviour**
No error response

**Console Output, Screenshots**
see above
**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.
pi 3b+
HQ camera
**Additional context**
Add any other context about the problem here.

chrisruk, 2023-02-06:
I'm just wondering if you might be using an older version of picamera2.

Do you mind pasting the output of 'pip show picamera2', it should show version 0.3.8.

johncblacker, 2023-02-07:
Shows version 0.3.8

johncblacker, 2023-02-08:
I ran a apt list --upgradeable
found libcamera was upgradeable; so I took a wile stab and did 
sudo apt-get upgrade
libcamera got upgraded and then the above code as well as other code I found in the documentation worked also (it didn't before I did the libcamera upgrade).  So, problem solved.

closed 2023-02-08 by johncblacker

navdotnetreqs, 2024-06-21:
pi@pi:~ $ pip show picamera2
Name: picamera2
Version: 0.3.19
Summary: The libcamera-based Python interface to Raspberry Pi cameras, based on the original Picamera library

Traceback (most recent call last):
  File "/home/pi/2stream.py", line 32, in <module>
    picam2.video_configuration.controls.FrameRate = 25.0
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'FrameRate'