raspberrypi/picamera2 #556: How to get started

baqwas, 2023-02-06:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
How to start using **picamera2**?

**Describe alternatives you've considered**
From [documentation](https://github.com/raspberrypi/picamera2/) I tried the following:
```
$ sudo apt install -y python3-pyqt5 python3-opengl
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
python3-opengl is already the newest version (3.1.5+dfsg-1).
python3-opengl set to manually installed.
python3-pyqt5 is already the newest version (5.15.2+dfsg-3).
python3-pyqt5 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
$ sudo apt install -y python3-picamera2
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
**python3-picamera2 is already the newest version (0.3.8-1).**
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
$ python
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import time
>>> from picamera2 import PiCamera2, Preview
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'PiCamera2' from 'picamera2' (/usr/lib/python3/dist-packages/picamera2/__init__.py)
>>> 

```

**Additional context**
OS: Debian GNU/Linux 11 (bullseye) aarch64 
Host: Raspberry Pi 3 Model B Rev 1.2 

As a complete _newbie_, how do I get started with **picamera2** self-paced exercises?

Thanks.

chrisruk, 2023-02-06:
It looks like you've used the wrong capitalisation in the import statement.

Which should be -
```
from picamera2 import Picamera2
```

https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf is very useful.

Along with the examples in this repository.

baqwas, 2023-02-06:
I become a bigger idiot everyday. Sorry to waste your time on something like this. I feel terrible because I can't even cut-and-paste correctly.

closed 2023-02-06 by baqwas