raspberrypi/picamera2 #557: [BUG] switch_mode_and_capture_image not work

jenyak, 2023-02-07:
**Describe the bug**
When I run the example [`capture_image_full_res.py`]( https://github.com/raspberrypi/picamera2/blob/5858f49b1e6628b236000a0191ecee448fe2c0d9/examples/capture_image_full_res.py), nothing happens and the program terminates with an error. On the other hand, the other example  `capture_full_res.py ` runs fine and without errors.

**To Reproduce**
To reproduce run example [capture_image_full_res.py]( https://github.com/raspberrypi/picamera2/blob/5858f49b1e6628b236000a0191ecee448fe2c0d9/examples/capture_image_full_res.py)

**Expected behaviour**
The captured image appears on the screen

**Console Output, Screenshots**
```
[0:05:52.977809685] [2045]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:05:53.006361849] [2049]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media0
[0:05:55.346589484] [2045]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-XBGR8888
[0:05:55.347054383] [2049]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
qt.qpa.xcb: QXcbConnection: XCB error: 148 (Unknown), sequence: 192, resource id: 0, major code: 140 (Unknown), minor code: 20
[0:05:58.037289315] [2053]  INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-BGR888
[0:05:58.041001211] [2049]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
[0:05:59.253430728] [2053]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-XBGR8888
[0:05:59.274243910] [2049]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Traceback (most recent call last):
  File "/home/pi/capture_image_full_res.py", line 19, in <module>
    image.show()
AttributeError: 'NoneType' object has no attribute 'show'
```

**Hardware :**
RPI 4b and HQ camera

**Additional context**
The problem might be that [switch_mode_and_capture_image](https://github.com/raspberrypi/picamera2/blob/5858f49b1e6628b236000a0191ecee448fe2c0d9/picamera2/picamera2.py#L1432)  is the only capture function that doesn't return anything

tvoverbeek, 2023-02-07:
Yes, that should be it. Try to add `return ` before the `self.dispatch.functions(...` on the last line of switch_mode_and_capture_image.
Now it returns nothing, so the image object is None, hence the attribute error.

davidplowman, 2023-02-13:
Thanks for spotting this and for the PR! I've replied further over there.

davidplowman, 2023-02-21:
We've merged this and should be making an updated release very soon, so I'll mark this one as closed. Thanks again.

PR #559 (2023-02-15): Fix `switch_mode_and_capture_image` method
Fixes #557 
Its looks like `return` was unintentionally omitted in [PR 436 ](https://github.com/raspberrypi/picamera2/commit/cff52ea865c954d6faa4e00f0003e16d7fc7d710#diff-26e92d0368f682e2f447e8e5c597e95d27f67eff88e4345d3d4fff4226eb8fcbR1449)
closed 2023-02-21 by davidplowman