raspberrypi/picamera2 #558: [BUG] Size 640x480  is not treated correctly for camera module v.2

neuralassembly, 2023-02-08:
In order to change the resolution of the preview, I modified examples/preview.py as follows.

```
picam2.start_preview(Preview.QTGL)
preview_config = picam2.create_preview_configuration()

->

(w, h)=(640, 480)
picam2.start_preview(Preview.QTGL, width=w, height=h)
preview_config = picam2.create_preview_configuration({'size': (w, h)})
```

Then, the preview window shows a very narrow area, as if it were cropped from a larger image (e.g. 1640x1232).

So, I think the size 640x480 is not treated correctly for the camera module v.2.
I noticed that examples/opencv_face_dectect.py has the same issue.

When I used the camera module v.1, the size 640x480 seemed to be treated correctly.

davidplowman, 2023-02-13:
Hi, it's true that selecting the correct camera mode is a surprisingly difficult thing to do automatically. Some people favour field of view, others framerate, so it's hard to guess. But you can force the selection of a particular camera mode by specifying a raw stream size, for example:
```
preview_config = picam2.create_preview_configuration(raw={'size': (1640, 1232)})
```
There more on this in section 4.2.2 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf). Hope that helps!

neuralassembly, 2023-02-15:
Thank you for your reply.

Based on the manual, I found that the following command returns satisfactory images for my purpose with camera module v.2.

```
preview_config = picam2.create_preview_configuration({'size': (640, 480)}, raw=picam2.sensor_modes[3])
```

I still cannot understand the correct choice of sensor modes, but I will try to read the manual in detail.

davidplowman, 2023-02-20:
Indeed I think this is a confusing area. If you don't say what mode you want then libcamera uses some fairly arbitrary heuristics, the result of which can be difficult to predict. Rather than try to understand those, I would say that a better strategy is to look at the list of modes your sensor has, and manually select the one you want (as you have done above).

davidplowman, 2023-02-28:
Hi again, if there's nothing else here for us to look at I'll close this issue shortly. Thanks!

davidplowman, 2023-03-09:
I'll probably close this one for now. Please open up a new issue if there's anything else. Thanks!

closed 2023-03-09 by davidplowman