raspberrypi/picamera2 #560: [OTHER] Picamera2 & Cronjob

Maddy1206, 2023-02-09:
Hello everyone!

I am trying to start Picamera2 with `crontab -e`. Unfortunately, I can't seem to get it working.
The preview is working and my scripts are also working and capturing the video storing the data to a file.

But whenever my script is invoked from cron I got the error: `Camera __init__ sequence did not complete.`

Does anybody has a clue?

davidplowman, 2023-02-13:
Hi, thanks for the question. I'm afraid I don't know the answer, but it feels like it might be related to permissions in some way. I expect others have faced this too, so if anyone does have a fix or workaround I'd be very pleased to know so that I can add it to the official manual.

davidplowman, 2023-02-13:
Could you say a bit more about how you're using crontab? Are you running the job as root or as a regular user? Are you trying to get something to run at startup, or once the system has been up for a while? These are all likely to affect what you can and can't do. Another factor may be environment variables - are there any that you're setting up that might not be picked up by a cron job?

Maddy1206, 2023-02-22:
I will come back to you in a few days! Sorry, I am not able to have a closer look at this project atm.

tvoverbeek, 2023-02-23:
I have seen this error when there was still another process using the camera running on another terminal, i,e, by not calling picamera2.stop()

Maddy1206, 2023-02-23:
@tvoverbeek That is true, normally. But the job doesn't even start after a fresh reboot while no instance of picamera2 has been used. In my scripts I also switch between different instances which does work, but only after starting it manually.
I did not find any solution myself nor in the web search, that's why I opened this ticket.

davidplowman, 2023-02-24:
My suspicion is that the error message is a bit generic, libcamera is saying "I couldn't open it so someone else must be using it". I think it may be more likely that there's some other reason, such as not having the right permissions, but I'm rather guessing here.

tvoverbeek, 2023-02-24:
@Maddy1206  Why are you running the cron job as root?

I just tried to run a video capture as a cron job and it works fine.
My default user is `tvo`, not `pi`.
Here is my crontab entry generated as user `tvo` with `crontab -e`
```
# m h  dom mon dow   command
10 22 * * * /home/tvo/mp4output.py >> /home/tvo/cron.log 2>&1
```
Note: stdout and stderr are redirected to `/home/tvo/cronlog` otherwise cron tries to mail the output which in my case fails because there is no MTA (Mail Transfer Agent) installed.

Here is the `mp4output.py` script:
```
#! /usr/bin/python3

#import libraries
import time
import os
import cv2
from picamera2 import MappedArray, Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FfmpegOutput

#set variables
hour = int(time.strftime('%H'))
now = time.strftime("%y_%m_%d_%H_%M")
h264_output = "test"+now+".h264"
mp4_output = "test"+now+".mp4"

#set timestamp
colour = (255, 255, 255)
origin = (0, 30)
font = cv2.FONT_HERSHEY_SIMPLEX
scale = 1
thickness = 2

def apply_timestamp(request):
    timestamp = time.strftime("%Y-%m-%d %X")
    with MappedArray(request, "main") as m:
        cv2.putText(m.array, timestamp, origin, font, scale, colour, thickness)

#set video configuration
picam2 = Picamera2()
picam2.pre_callback = apply_timestamp
picam2.video_configuration.controls.FrameRate = 25.0
picam2.video_configuration.size = (1280, 720)
encoder = H264Encoder(bitrate=10000000)
output = FfmpegOutput(mp4_output)

#set length of footage
picam2.start_recording(encoder, output)
time.sleep(10)
picam2.stop_recording()
```
The log file for the run at 22;10:
```
[6:41:41.393397975] [1713]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[6:41:41.416827658] [1714]  WARN CameraSensorProperties camera_sensor_properties.cpp:205 No static properties available for 'imx708_wide_noir'
[6:41:41.416940210] [1714]  WARN CameraSensorProperties camera_sensor_properties.cpp:207 Please consider updating the camera sensor properties database
[6:41:41.529705792] [1714]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx708@1a to Unicam device /dev/media3 and ISP device /dev/media0
[6:41:41.538514846] [1713]  INFO Camera camera.cpp:1028 configuring streams: (0) 1280x720-XBGR8888
[6:41:41.539169998] [1714]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx708@1a - Selected sensor format: 1536x864-SBGGR10_1X10 - Selected unicam format: 1536x864-pBAA
```
and the entry in /var/log/syslog:
```
Feb 24 22:10:01 picam2-32 CRON[1712]: (tvo) CMD (/home/tvo/mp4output.py >> /home/tvo/cron.log 2>&1)
```

Maddy1206, 2023-03-02:
@tvoverbeek I was running it on root because of the assumption that permissions for the Picamera were denied.

Anyway, I found out that my script file was showing unusal behavior because I created new files and with them the cronjob worked. 
And just to make sure I did a full reset and installed a new OS. After that everything worked fine.

Thanks a lot! I am closing this issue.

closed 2023-03-02 by Maddy1206