raspberrypi/picamera2 #561: [BUG] RPI4 on latest raspios bullseye lite arm64 missing libcamera.so.0 after python3-picamera2 install

pawelzebrowski, 2023-02-11:
**Please only report one bug per issue!**

**Describe the bug**
Freshly installed Raspios Bullseye Lite (https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-26/) with python3-picamera2 installed appears to be missing libcamera.so.0

**To Reproduce**

1. Install https://downloads.raspberrypi.org/raspios_lite_arm64/images/raspios_lite_arm64-2022-09-26/
2. `sudo apt-get -y install i2c-tools python3-picamera2`
3. `libcamera-hello`

**Expected behaviour**
`libcamera-hello` does its thing

**Console Output, Screenshots**
`libcamera-hello: error while loading shared libraries: libcamera.so.0: cannot open shared object file: No such file or directory`

**Hardware :**
RPI4 with IMX219

This same hardware works fine when installed using
```
sudo apt-get -y python3-libcamera
sudo pip install picamera2
```

**Additional context**
```
bash# ldconfig -v | grep camera
ldconfig: Can't stat /usr/local/lib/aarch64-linux-gnu: No such file or directory
ldconfig: Path `/usr/lib/aarch64-linux-gnu' given more than once
ldconfig: Path `/lib/aarch64-linux-gnu' given more than once
ldconfig: Path `/usr/lib/aarch64-linux-gnu' given more than once
ldconfig: Path `/usr/lib' given more than once
	libcamera.so.0.0.3 -> libcamera.so.0.0.3
	libcamera-base.so.0.0.3 -> libcamera-base.so.0.0.3
	libcamera_app.so -> libcamera_app.so
ldconfig: /lib/aarch64-linux-gnu/ld-2.31.so is the dynamic linker, ignoring
```

```
bash# sudo find / -name *libcamera*.so |& grep -v denied
/usr/lib/python3/dist-packages/libcamera/_libcamera.cpython-39-aarch64-linux-gnu.so
/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstlibcamera.so
/usr/lib/aarch64-linux-gnu/libcamera_app.so
```

davidplowman, 2023-02-13:
Hi, libcamera is still undergoing developement so there are occasions when various packages have to be updated together. Could you just try a `sudo apt update` and `sudo apt upgrade` and see if that resoves the problem? Thanks!

pawelzebrowski, 2023-02-13:
That did the trick, thanks!

closed 2023-02-13 by pawelzebrowski