raspberrypi/picamera2 #562: Problems with frame server thread and picam2 thread not stopping

johncblacker, 2023-02-12:
I have a python web app built using the flask server.  In my "main" thread at startup I start the frame server (from the examples dir), then picam2, then another thread that takes the frames from the frame server and uses motion detection algorithm on them in order to draw a rectangle around the area where motion is detected and displays the frame.  The app also allows the user to change various camera settings (brightness, contrast, etc.) and the frame size.  When a frame size change is requested the app attempts to stop the frame server thread, the picam2 object and the motion-detection thread.  My problem is that the frameserver.stop(), picam2.stop() and the motion-detection thread don't terminate.  To terminate the motion-detection thread I use thrmotion.join().  Even if I issue frameserver.stop(), picam2.stop() and thrmotion.join() in my signal handler, the threads never end and the app won't terminate.  All the threads for the app show as "daemon" threads when I list them out with threading._active.items()
So the question is: why wont the threads stop?  I can't change the "size" of the image without stopping and starting the picam2 object with a new "size" parameter, thus the need for stopping the threads.  I'm trying to do this without requiring the use to terminate their browser session and restart it - which would be an option if I were to "persist" the size change in a config file and then use the new size value when starting the app again.

I've perused the python doc and various threads from a web search for: "python threads not stopping" to no avail.
Thanks.

davidplowman, 2023-02-13:
Hi, it sounds like things _should_ work if you stop the frame server clients first (that's the motion detection thread), then stop the frame server, and then you can stop and reconfigure the picam2 object. But if that's what you're doing then presumably something must be going wrong.

Are you able to distil the failure case into a minimal example that shows the problem - the simpler the better, of course! But then we'd have something we could try out and investigate for you. Thanks!

johncblacker, 2023-02-13:
Does it matter that both frameserver and picam2 threads are daemon threads?  I thought I read somewhere online daemon threads couldn't be stopped. But in my signal handler they are...

johncblacker, 2023-02-13:
I will try to cobble a minimal app for testing.

johncblacker, 2023-02-13:
Problem solved!  Created a mini-app with only the camera reset code on the webpage and the reset function in my app.py.
I had created a variable in the FrameServer class called thrfrsthread which I had populated by an assign statement in the "start()"
function, after the self._thread.start(); however, I had populated it by issuing a "self.thrfrsthread = threading.current_thread()" and as it turns out the value was the "Main Thread."  I changed that in the "__init__" function to: 
self.thrfrsthread = self._thread and I loaded that value in my camera-reset function of the app with:
thrfrsthread = frserver.thrfrsthread
Now I have the correct thread in my app for the frserver; and, I can test  via:
while (thrfrsthresd.is_alive):
Then, to verify that the picam2.stop() indeed worked I changed that to picam2.stop_() and it returns a tuple the first
one of which is True when the camera stops and the second is None.  So, I can simply check the first value of the tuple
for True to verify the camera has been stopped.
Anyway, I believe I'm on my way, although I haven't gone all the way through the restart process yet.  Your suggestion of creating a mini-app was a good one and I thank you for that.  Don't know why the app was hanging (and I thought the threads werren't ending) but it was probably that I wasn't getting the motion-detect thread to end and that was a show-stopper.

johncblacker, 2023-02-13:
Closed, user error, if I didn't use it I wouldn't get the error?

closed 2023-02-13 by johncblacker

davidplowman, 2023-02-14:
Great, glad that you've been able to resolve it!

johncblacker, 2023-02-14:
As luck would have it, when I integrated changes from mini-app back to original, it stopped working...so, still have an issue to resolve. Maybe I missed something because motion detect thread isn't  ending.

⁣Get BlueMail for Android ​

johncblacker, 2023-02-14:
I give up!  I stopped using FrameServer altogether because I don't need it.  But, even then after I issue picam2.stop() and/or
picam2.stop_() and then try to restart the camera again I get errors during initialization.  So, I'm abandoning being able to change the frame size (size) parameter which requires the restart process, stopping threads (which don't seem to stop), etc.   I did make a change to picamera2 for my own use (in dist_packages) and added: "self.thread.name = "Picamera2" in the CameraManager setup() function.  At least that lets me know what thread I'm dealing with.

johncblacker, 2023-02-14:
I propose a change to picamera2.py module:  add: self.thread.name = ‘Picamera2’

to the CameraManager setup() function so people like me know which thread is for the camera.

 

I’ve removed the frame server from my app – really don’t need the extra complexity of another thread; but still I’ve run into

a problem stopping the camera, then restarting it again.  I seem to get it stopped via picam2.stop or picam2.stop_() but when 

I try to get it started again, I get a failure during the init because apparently it hasn’t really stopped and released the resources 

so it won’t start again.

 

Thanks for you help!

John

davidplowman, 2023-02-15:
Yes, I can change the thread name (or please submit a PR if you'd like to do it yourself!).

I wouldn't have expected re-starting after stopping to be a problem, because stopping doesn't free the camera resources. But there might be issues if you try to re-acquire the camera (by making a new Picamera2 object, for example). As ever, if you have a simple test case then we can investigate.

In general, invoking camera operations from within the camera thread is a rather dangerous thing to do - there's some stuff on this in section 8.5 of the [manual](https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf), though it's somewhat specific to Qt applications where we anticipate it being an issue.

johncblacker, 2023-02-15:
Sorry, but I'm not experienced enough with github to be able to do the "pr" process, so I'd appreciate your doing it.  I just did some research on the "pr" process and don't think I know it well enough to tackle it...I've got enough on my plate just getting this app to work; which brings me to: I changed from trying to create another picam2 object and instead just using a picam2.stop() and then a configure and a picam2.start().  After the picam2.start() I get what appears to be normal messages indicating that the picam2.start() has worked, however, I have a print statement after the picam2.start() and it doesn't hit.  It's like something is still waiting (main thread?) for some reason.  Consequently I cannot start the motion-detector thread back up and the app is basically dead.  What do you think of this: instead of terminating the motion-detection thread I just put a "pause" capability in it such that if the pause variable is set, the motiont-detect function simply continues but does not issue an command to get image via picam2.capture_array().  I'm thinking that might resolve this last issue, but I won't know until I give it a shot - which is easy to do.

davidplowman, 2023-02-15:
I'll take care of that thread name for the next release, no problem.

From your explanation it sounds like there's something that just isn't behaving like it should. Threads shouldn't lock up and so on unless something more fundamental is wrong, and in those circumstances it's hard to say what would work (if anything, in fact). I'm still inclined to the view that recreating the problem in a test application would enable us to figure out what is actually happening.