raspberrypi/picamera2 #563: [HOW-TO] Ubuntu 22.04 Install

brianabouchard, 2023-02-13:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**
I would like to use picamera2 on Ubuntu 22.04. I've been able to manually build and install the libcamera library and all of the other dependencies, but I am now stuck on python3-kms++. Ubuntu 22.04 uses Python 3.10, while this library appears to use 3.9 and it will not install. Is there any way that I can build this package from source? I've been unable to find any information on it online.

davidplowman, 2023-02-13:
Hi, I'm thinking it ought to be possible to build python3-kms++ on a Ubuntu system so that it has the correct Python version, would that be correct? We can't make a package for Ubuntu, but maybe it's worth asking them about it. In any case it's a relatively small package and I don't recall any particular problems building it locally, so that might be a solution. Sorry not to have a ready-made answer!

brianabouchard, 2023-02-13:
That would be fantastic, I'd be happy to build it locally. Do you have a link to the docs or any info on this package? I couldn't seem to find the right repo.

davidplowman, 2023-02-14:
Hi, the repository in question is this one: https://github.com/tomba/kmsxx (I agree, it is a bit obscure to find!!)

KhanMechAI, 2023-02-15:
I would also like to build from source for a docker container on raspberry pi 4B

KhanMechAI, 2023-02-15:
Main issue I've run into is using the pip installler - the cmake pypi package is completely broken. Its a dependency of simplejpeg and simplejpeg is a dependency of picamera2. Love playing this game...Tried may things to resolve this including building different versions of the pypi cmake (no success).

waveform80, 2023-04-25:
Getting `picamera2` into Ubuntu is on my to-do list for the mantic release (it was on my "if there's time" list for lunar, but I wound up burning too much time on module enablement amongst other things), so I'll stick this bug in my bookmarks and try and keep people updated here.

Incidentally, the `libcamera-apps` package should also be making it in (that one is basically ready, but missed the feature-freeze for lunar); `picamera2` looks to be a bit more complex with the extra dependencies but we'll see how it goes.

I wouldn't like to speculate on whether any work on this would get back-ported to jammy (22.04) though. Given all the dependencies, it would be a fairly major task.

Jai-GAY, 2024-01-12:
Ubuntu [22.04 64-bit release](https://ubuntu.com/about/release-cycle#ubuntu) has been for quite some time. To get [ROS2](https://www.ros.org/reps/rep-2000.html#humble-hawksbill-may-2022-may-2027), OpenCV, Pi Camera [into one system](https://www.youtube.com/live/OIqkmgOY_Ws?feature=shared&t=90) is a [huge challenge](https://github.com/raspberrypi/rpicam-apps/issues/628).

nbaldy, 2024-03-05:
Someone from this group might already be on it, but it looks like there's a prerelease version of python-libcamera. https://pypi.org/project/rpi-libcamera/. I'm testing it out now and will report back

davidplowman, 2024-03-06:
In case you haven't spotted it, there is information on rpi-libcamera [here](https://forums.raspberrypi.com/viewtopic.php?t=361758).

nbaldy, 2024-03-06:
Following up - those instructions above worked for me. It took some install twiddling to get all the right versions installed, but I got there. This was my total process, I think. 

### Install libcamera
```
sudo apt install -y libcamera-dev libepoxy-dev libjpeg-dev libtiff5-dev libpng-dev python3 python3-dev
sudo apt install libcamera-tools g++
sudo apt install -y python3-pip git python3-jinja2
sudo apt install -y libboost-dev
sudo apt install -y libgnutls28-dev openssl libtiff5-dev pybind11-dev liblttng-ust-dev
sudo apt install -y qtbase5-dev libqt5core5a libqt5gui5 libqt5widgets5
# NOTE - for Ubuntu 22.04, need to install meson via pip because the version apt gets is too old
sudo apt install -y cmake ninja-build 
sudo apt install -y python3-yaml python3-ply python3-pip libyaml-dev
sudo pip3 install --user meson # needs sudo to be on sudo pythonpath
PATH=$PATH:/home/$USER/.local/bin/
sudo apt install -y libglib2.0-dev libgstreamer-plugins-base1.0-dev

cd 
git clone https://github.com/raspberrypi/libcamera.git
cd libcamera

meson setup build --buildtype=release -Dpipelines=rpi/vc4,rpi/pisp -Dipas=rpi/vc4,rpi/pisp -Dv4l2=true -Dgstreamer=enabled -Dtest=false -Dlc-compliance=disabled -Dcam=disabled -Dqcam=disabled -Ddocumentation=disabled -Dpycamera=enabled

ninja -C build   # use -j 2 on Raspberry Pi 3 or earlier devices
sudo ninja -C build install
```

### Install rpicam-apps
```
sudo apt install -y libboost-program-options-dev libdrm-dev libexif-dev
cd 
git clone https://github.com/raspberrypi/rpicam-apps.git --branch v1.4.3
cd rpicam-apps

# Choose the 'lite' option
meson setup build -Denable_libav=false -Denable_drm=true -Denable_egl=false -Denable_qt=false -Denable_opencv=false -Denable_tflite=false

meson compile -C build # use -j1 on Raspberry Pi 3 or earlier devices
sudo /home/$USER/.local/bin/meson install -C build
sudo ldconfig # this is only necessary on the first build
```

### Install  libkms dependencies
```
sudo apt update && sudo apt upgrade
sudo apt install -y  libfmt-dev libdrm-dev libfmt-dev
# Install kmsxx
git clone https://github.com/tomba/kmsxx.git
cd kmsxx/
git submodule update --init
meson build -Dpykms=enabled
sudo ninja -C build install
```

### Install picamera
```
 pip install rpi-libcamera rpi-kms picamera2
```

From there, I could import and use picamera. 

Note, I'm still having some minor issues with my camera which are at the libcamera driver layer (and possibly hardware) so I cannot 100% say it works to completion... but this is my best set of installs at the moment, (before falling back to RPI OS to test if my assumedly unrelated issue with the camera is a driver issue)

allison-truhlar, 2024-09-14:
Thanks so much for these detailed instructions, @nbaldy! This is the only method that's worked for me to get picamera2 installed inside an Ubuntu 22.04 Docker container (built from the ros2:humble-perception image). The only extra step I had to add was an `apt install libcap-dev` prior to the `pip install rpi-libcamera rpi-kms picamera2`. 

The issue I have now is the camera can't seem to be found inside the Docker container. I found these steps I will try to fix this problem: https://www.losant.com/blog/how-to-access-the-raspberry-pi-camera-in-docker.

waveform80, 2025-02-05:
Quick update on where we are in Ubuntu:

Well, this didn't happen in mantic, or noble, or oracular (long story, but the last year has not been a pleasant one). However, I've now got some help and we're reasonably confident we'll be getting libcamera at least working in plucky (25.04).

However, getting picamera2 on top of that is going to prove tricky. There's a whole pile of dependencies we don't have in the archive yet (simplejpeg, kms++, etc.) and some of those are already [broken by the bump to numpy 2.x](https://bugs.launchpad.net/ubuntu/+bug/2091974).

So ... some progress, but a ways to go yet.

fatemeakbari, 2025-03-31:
@nbaldy 
Could you please share the exact versions of the libraries you have installed?
When I installed rpi-camera, I got an incompatible version message of libcamera

allison-truhlar, 2025-04-03:
@fatemeakbari 

Here are the library versions that are working for me (code from a Dockerfile):

```
# libcamera
WORKDIR /usr/local/
RUN git clone https://github.com/raspberrypi/libcamera.git --branch v0.3.1+rpt20240906
WORKDIR /usr/local/libcamera/
RUN meson setup build --buildtype=release \
    -Dpipelines=rpi/vc4,rpi/pisp \
    -Dipas=rpi/vc4,rpi/pisp \
    -Dv4l2=true \
    -Dgstreamer=enabled \
    -Dtest=false \
    -Dlc-compliance=disabled \
    -Dcam=disabled \
    -Dqcam=disabled \
    -Ddocumentation=disabled \
    -Dpycamera=enabled && \
    ninja -C build && \
    sudo ninja -C build install

# rpicam-apps
WORKDIR /usr/local/
RUN git clone https://github.com/raspberrypi/rpicam-apps.git --branch v1.5.1 
WORKDIR /usr/local/rpicam-apps/
RUN meson setup build \
    -Denable_libav=enabled \
    -Denable_drm=enabled \
    -Denable_egl=enabled \
    -Denable_qt=enabled \
    -Denable_opencv=disabled \
    -Denable_tflite=disabled
RUN meson compile -C build
RUN meson install -C build
RUN sudo ldconfig

# kmsxx
WORKDIR /usr/local/
RUN git clone https://github.com/tomba/kmsxx.git
WORKDIR /usr/local/kmsxx
RUN git reset --hard 6cf6e88715ac034f568603bce9a1b8f4a30c12ce
RUN git submodule update --init && \
    meson build -Dpykms=enabled && \
    sudo ninja -C build install

# picamera2 and rpi-libcamera
# upgrade pip required to passed setup-args
RUN pip install --upgrade pip
RUN pip install picamera2==0.3.21
RUN pip install rpi-libcamera==0.1a6 -C setup-args="-Drevision=v0.3.1+rpt20240906"
```

fatemeakbari, 2025-04-04:
thanks alot @allison-truhlar 
But I still have a problem. When I tried to install rpi-libcamera, I encountered the following error:

```
$ rpicam-still --version
rpicam-apps build: d7a1a13b041e-intree 04-04-2025 (06:00:21)
libcamera build: v0.3.1+50-69a894c4
```

```
$ pip install rpi-libcamera==0.1a6 -C setup-args="-Drevision=v0.3.1+rpt20240906"
Collecting rpi-libcamera==0.1a6
  Downloading rpi_libcamera-0.1a6.tar.gz (4.8 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Installing backend dependencies ... done
  Preparing metadata (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Preparing metadata (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [22 lines of output]
      + meson setup /tmp/pip-install-c67_3grm/rpi-libcamera_9224ccad1ad14c448657edaff4b5c6dd /tmp/pip-install-c67_3grm/rpi-libcamera_9224ccad1ad14c448657edaff4b5c6dd/.mesonpy-ccp2s2uf -Dbuildtype=release -Db_ndebug=if-release -Db_vscrt=md -Drevision=v0.3.1+rpt20240906 -Drepository=https://github.com/raspberrypi/libcamera.git -Dversion=v0.3.1+50-69a894c4 -Drevision=v0.3.1+rpt20240906 --native-file=/tmp/pip-install-c67_3grm/rpi-libcamera_9224ccad1ad14c448657edaff4b5c6dd/.mesonpy-ccp2s2uf/meson-python-native-file.ini
      The Meson build system
      Version: 1.7.1
      Source dir: /tmp/pip-install-c67_3grm/rpi-libcamera_9224ccad1ad14c448657edaff4b5c6dd
      Build dir: /tmp/pip-install-c67_3grm/rpi-libcamera_9224ccad1ad14c448657edaff4b5c6dd/.mesonpy-ccp2s2uf
      Build type: native build
      Project name: pylibcamera
      Project version: 0.1.0
      C compiler for the host machine: cc (gcc 13.3.0 "cc (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
      C linker for the host machine: cc ld.bfd 2.42
      C++ compiler for the host machine: c++ (gcc 13.3.0 "c++ (Ubuntu 13.3.0-6ubuntu2~24.04) 13.3.0")
      C++ linker for the host machine: c++ ld.bfd 2.42
      Host machine cpu family: aarch64
      Host machine cpu: aarch64
      Message: Cloning from: https://github.com/raspberrypi/libcamera.git, revision v0.3.1+rpt20240906
      Message: libcamera version is v0.2.0
      
      
      ../meson.build:26:8: ERROR: Problem encountered: This package works with libcamera version v0.3.1+50-69a894c4, but you have version v0.2.0
       installed. Please check the readme for help with this error
      
      A full log can be found at /tmp/pip-install-c67_3grm/rpi-libcamera_9224ccad1ad14c448657edaff4b5c6dd/.mesonpy-ccp2s2uf/meson-logs/meson-log.txt
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: metadata-generation-failed

× Encountered error while generating package metadata.
╰─> See above for output.

note: This is an issue with the package mentioned above, not pip.
hint: See above for details.

```

allison-truhlar, 2025-04-07:
@fatemeakbari A couple of questions - 

- Did you follow all the steps as I wrote them, in a fresh environment, prior to trying to run the rpi-libcamera install?
- Is there an existing version of libcamera installed in your environment, prior to running the other install?