raspberrypi/picamera2 #565: [BUG] JpegEncoder() fails on with YUV420 data

ballen4705, 2023-02-14:
RELEASE:
picamera2 0.3.8

SUMMARY:
The documentation and source code imply that JpegEncoder() can be used with YUV420 data.  However there is a problem in the matching of array dimensions:  simplejpeg._jpeg.encode_jpeg() expects a 3-index array, but JpegEncoder() in picamera2 is passing it a 2-index array.

REPRODUCE:
(1) start with the example program mjpeg_server.py 

(2) change two lines:

    picam2.configure(picam2.create_video_configuration(main={"size": (640, 480), "format":"YUV420"}))

    picam2.start_recording(JpegEncoder(colour_subsampling="420"), FileOutput(output))

    Output pasted below.

FIX:
To fix this it might be enough to add a third "dummy" index x = x[:,:,np.newaxis] or equivalently x = x[:,:,None] for YUV420.

NOTE:
I have tried various options for the colour_space argument of JpegEncoder.  But from reading the source of simplejpeg._jpeg.encode_jpeg() it doesn't appear that this provides an option for a 2d input array.

OUTPUT:
ballen@rp:~ $ python mjpeg_server_with_annotate_callback.py
[5:33:08.791122833] [4259]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[5:33:08.822459122] [4260]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media1 and ISP device /dev/media2
[5:33:08.827267284] [4259]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-YUV420
[5:33:08.827643228] [4260]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/multi_encoder.py", line 46, in output_thread
    buffer, timestamp_us = task.result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 440, in result
    return self.__get_result()
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 389, in __get_result
    raise self._exception
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/multi_encoder.py", line 58, in do_encode
    buffer = self.encode_func(request, request.picam2.encode_stream_name)
  File "/usr/lib/python3/dist-packages/picamera2/encoders/jpeg_encoder.py", line 41, in encode_func
    return simplejpeg.encode_jpeg(array, quality=self.q, colorspace=self.colour_space,
  File "simplejpeg/_jpeg.pyx", line 410, in simplejpeg._jpeg.encode_jpeg
ValueError: Buffer has wrong number of dimensions (expected 3, got 2)

davidplowman, 2023-02-14:
Hi, there are some slightly confusing things here. Let me try to explain.

Firstly, the JpegEncoder only accepts RGB pixel formats, not YUV ones so what you're trying to do can't work. In fact, it's very common (and annoying, I might add) with Python that YUV formats generally receive minimal or no support at all

Secondly the colour_subsampling parameter refers to the output, not the input (which is of course a standard interleaved RGB type format). It determines what colour subsampling happens internally when the RGB is converted to YUV for JPEG encoding. So this determines what a JPEG decoder would report when you ask it about the encoded image.

Obviously it's a bit disappointing that you can't give it YUV420 directly, and instead it goes to great expense to do this conversion internally for you when you never wanted it in the first place! But you do also have the option of using the hardware MJPEGEncoder which should accept YUV420 directly.

I hope that's a bit clearer. I'll make some improvements to the documentation.

ballen4705, 2023-02-14:
Hi David,

thanks for the quick and helpful response.  A few clear sentences in the documentation would go a long way to help others to avoid this pitfall.

Is there an example script that feeds YUV420 to the hardware MJPEGEncoder?   Google found this, which looks like a potentially good starting point: https://github.com/raspberrypi/picamera2/issues/473

I'll respond to your Q&A post later today.

Cheers,
    Bruce

davidplowman, 2023-02-14:
I've made some documentation updates. Of course it'll be "a few days" before they appear.

In the meantime, [this](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_mjpeg_v4l2.py) is not a bad starting point, though I note it doesn't explicitly request YUV420 as the format. But that should be a simple change to the format requested in the configuration.

ballen4705, 2023-02-14:
Thank you,  I'll give this a try:

```
<Insert near the bottom of examples/mjpeg_server.py>

picam2 = Picamera2()
video_config = picam2.create_video_configuration(main={"size": (640, 480),"format":"YUV420"})
picam2.configure(video_config)
encoder = MJPEGEncoder(10000000)
output = StreamingOutput()
picam2.start_recording(encoder, output)

try:
    address = ('', 8000)
    server = StreamingServer(address, StreamingHandler)
    server.serve_forever()
finally:
    picam2.stop_recording()

```

davidplowman, 2023-02-15:
Great, glad it's working. I think I can probably close this one now!

closed 2023-02-15 by davidplowman