raspberrypi/picamera2 #566: How to change the FPS with picamera2 library ?

aguinand, 2023-02-14:
Hello, i am experiencing issues with picamera2 regarding the FPS. I am able to change absolutely everything, it's fantastic. Size, bitrate etc.
But absolutely impossible to have the FPS set to 25. There are 30 frames per second set as default, but i need 25 for my project.
It works perfectly with libcamera, so i guess it should work with picamera2.

When i run the python code with this command:

picam2.video_configuration.controls.FrameRate = 25.0

I get this error:

Traceback (most recent call last):
  File "/home/pi/openturtlecam.py", line 30, in <module>
    picam2.video_configuration.framerate = 25.0
  File "/usr/lib/python3/dist-packages/picamera2/configuration.py", line 46, in __setattr__
    raise RuntimeError(f"Invalid field '{name}'")
RuntimeError: Invalid field 'framerate'

Thanks for your help !

davidplowman, 2023-02-14:
Hi, the line quoted in the error message (`picam2.video_configuration.framerate = 25.0`) is definitely incorrect, but is not the same as the line you quoted earlier (`picam2.video_configuration.controls.FrameRate = 25.0`) which does look correct. Probably best to clear that up first. Thanks!

aguinand, 2023-02-14:
Yes sorry, I went too fast (as usual). Corrected:

When i run this command (that i found here https://datasheets.raspberrypi.com/camera/picamera2-manual.pdf):

picam2.video_configuration.controls.FrameRate = 25.0

I get this error:

Traceback (most recent call last):
  File "/home/pi/remoratimestamp.py", line 33, in <module>
    picam2.video_configuration.controls.FrameRate = 25.0
AttributeError: 'dict' object has no attribute 'FrameRate'

aguinand, 2023-02-14:
I did some more testing, and I managed to get rid of the error thanks to your 

from picamera2 import Picamera2
picam2 = Picamera2()
picam2.video_configuration.controls.FrameRate = 25.0

Unfortunately, my footage still displays the default 30fps after I run the script (and no errors).

davidplowman, 2023-02-14:
Exactly how are you recording the footage (and in what format), and how are you determining that it's 30fps? Remember that h.264 bitstreams don't have proper timestamps in them so that could be an issue (though doubtless there could be other causes too).

aguinand, 2023-02-14:
I am recording the footage with the Raspberry camera module 3 wide.

At first, I was doing tests with libcamera, where I managed to have 25fps footage, 1920x1080p. Once the .h264 file is recorded, I convert with MP4box. Then, on my computer, i am able with VLC or Quicktime to check the encoded FPS of the footage. So, to resume, with libcamera I can easily record 25fps footage.
But when I use picamera2 library with python, i can't seem to change the default 30fps even though all the other settings work perfectly. With the footage done through python, i also convert the h264 to mp4. I spend hours trying to narrow down the possible sources of, but I always end up with the picamera2 library.

aguinand, 2023-02-15:
It worked ! I used FFMPEG instead of MP4box as you suggested and I finally have a video with 25fps. Thank you so much ! Where do you live ? ^^

Here is my fully functional script, do you think it is worth putting it in the examples ?

`#import libraries
import time
import os
import cv2
from picamera2 import MappedArray, Picamera2
from picamera2.encoders import H264Encoder

#set variables
hour = int(time.strftime('%H'))
now = time.strftime("%y_%m_%d_%H_%M")
h264_output = "test"+now+".h264"
mp4_output = "test"+now+".mp4"

#set timestamp
colour = (255, 255, 255)
origin = (0, 30)
font = cv2.FONT_HERSHEY_SIMPLEX
scale = 1
thickness = 2

def apply_timestamp(request):
    timestamp = time.strftime("%Y-%m-%d %X")
    with MappedArray(request, "main") as m:
        cv2.putText(m.array, timestamp, origin, font, scale, colour, thickness)

#set video configuration
picam2 = Picamera2()
picam2.pre_callback = apply_timestamp
picam2.video_configuration.controls.FrameRate = 25.0
picam2.video_configuration.size = (1920, 1080)
encoder = H264Encoder(bitrate=17000000)

#set length of footage in seconds
picam2.start_recording(encoder, h264_output)
time.sleep(10)
picam2.stop_recording()

#convert h264 to mp4, and delete h264 file
os.system("ffmpeg -r 25 -i "+h264_output+" -c copy "+mp4_output)
os.remove(h264_output)

`

davidplowman, 2023-02-15:
It might be worth adding. I'm wondering whether there's a way to do it "as you go" using the `FfmpegOutput`  (see [here](https://github.com/raspberrypi/picamera2/blob/main/examples/mp4_capture.py)). It might be as simple as changing the filename to `"-r 25 test.mp4"` though I note it normally resamples the frame times on the way into FFmpeg so it might not work...

aguinand, 2023-02-15:
Yes, even better ! I now output directly to mp4 without going through h264:
```

#import libraries
import time
import os
import cv2
from picamera2 import MappedArray, Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FfmpegOutput

#set variables
hour = int(time.strftime('%H'))
now = time.strftime("%y_%m_%d_%H_%M")
h264_output = "test"+now+".h264"
mp4_output = "test"+now+".mp4"

#set timestamp
colour = (255, 255, 255)
origin = (0, 30)
font = cv2.FONT_HERSHEY_SIMPLEX
scale = 1
thickness = 2

def apply_timestamp(request):
    timestamp = time.strftime("%Y-%m-%d %X")
    with MappedArray(request, "main") as m:
        cv2.putText(m.array, timestamp, origin, font, scale, colour, thickness)

#set video configuration
picam2 = Picamera2()
picam2.pre_callback = apply_timestamp
picam2.video_configuration.controls.FrameRate = 25.0
picam2.video_configuration.size = (1920, 1080)
encoder = H264Encoder(bitrate=17000000)
output = FfmpegOutput(mp4_output)

#set length of footage
picam2.start_recording(encoder, output)
time.sleep(10)
picam2.stop_recording()
```

davidplowman, 2023-02-15:
The only thing to be aware of is that FFmpeg is resampling the times when it gets the frames. So this means there'll be some amount of jitter in them due to processing delays. So the frame-to-frame times will go up and down a bit, though the average will of course match the camera's exactly.

closed 2023-02-15 by aguinand

NadiaColodro, 2024-10-07:
Hi! I’m facing a similar issue. I was able to change the FPS, but the video duration becomes inaccurate when I convert it to MP4. Do you have any suggestions on how to fix this?

davidplowman, 2024-10-08:
If you use the `FfmpegOutput` you should get (reasonably) accurate timestamps. Alternatively you can save ah H.264 file along with a timestamps file (there's a `pts` argument to `start_encoder`), and then use a 3rd party tool (maybe `makemkv`?) to convert to an mp4.