raspberrypi/picamera2 #574: [HOW-TO] How to use realtime object detection example with other models (e.g. with EfficientDet-Lite)

arcsi8, 2023-02-18:
Hi guys I would like to know how can I use the real_time_with_labels.py with other models.
I have an EfficientDet-Lite0 model which works fine since I could test it on pictures but when I would like to use it
with the realtime example (--model mymodel.tflite) It throws this error: 

```
Traceback (most recent call last):
  File "/home/miscope/picamera2/examples/tensorflow/real_time_with_labels.py", line 154, in <module>
    main()
  File "/home/miscope/picamera2/examples/tensorflow/real_time_with_labels.py", line 150, in main
    _ = InferenceTensorFlow(grey, args.model, output_file, label_file)
  File "/home/miscope/picamera2/examples/tensorflow/real_time_with_labels.py", line 101, in InferenceTensorFlow
    for i in range(int(detected_boxes)):
TypeError: only size-1 arrays can be converted to Python scalars
```
This occurs almost immediately after the preview window had became visible.

```
 detected_boxes = interpreter.get_tensor(output_details[0]['index'])
    detected_classes = interpreter.get_tensor(output_details[1]['index'])
    detected_scores = interpreter.get_tensor(output_details[2]['index'])
    num_boxes = interpreter.get_tensor(output_details[3]['index'])

    rectangles = []
    for i in range(int(num_boxes)): #throws an error here
        top, left, bottom, right = detected_boxes[0][i]
        classId = int(detected_classes[0][i])
        score = detected_scores[0][i]
        if score > 0.5:
            xmin = left * initial_w
            ymin = bottom * initial_h
            xmax = right * initial_w
            ymax = top * initial_h
            box = [xmin, ymin, xmax, ymax]
            rectangles.append(box)
            if labels:
                print(labels[classId], 'score = ', score)
                rectangles[-1].append(labels[classId])
            else:
                print('score = ', score)

```
I also asked this on StackOverflow they told me , I should try this:
`for i in range(int(num_boxes[]0[0])): `

After this, the preview window did not close, and I did not get any error message. But for some reason
it's not detecting anything. And the program is not entering into the iterations:

```
 rectangles = []
    for i in range(int(num_boxes[0][0])): 
        top, left, bottom, right = detected_boxes[0][i]
        classId = int(detected_classes[0][i])
        score = detected_scores[0][i]
        print("HelloWorld") #to check whether it is entering or not 
        if score > 0.5:
            xmin = left * initial_w
            ymin = bottom * initial_h
            xmax = right * initial_w
            ymax = top * initial_h
            box = [xmin, ymin, xmax, ymax]
            rectangles.append(box)
            if labels:
                print(labels[classId], 'score = ', score)
                rectangles[-1].append(labels[classId])
            else:
                print('score = ', score)
```
But when I capture the object which should be recognized, the program does not print it on the terminal,

If you need any additional info let me know, I hope you can help.

davidplowman, 2023-02-20:
Hi, I'm afraid I don't really know anything about this, but I might start by printing out and comparing the actual data structures `num_boxes` and `detected_boxes` using the original model (that works), and also with your new one. What do they look like, and what's different between them? Please feel free to post some examples here if you'd like to discuss them. Sorry not to be more help!

davidplowman, 2023-02-27:
Is there anything else to look at here? If not I'll consider closing this issue in a few days. Thanks.

davidplowman, 2023-03-09:
I'll probably close this one for now. Please open up a new issue if there's anything else. Thanks!

closed 2023-03-09 by davidplowman