raspberrypi/picamera2 #576: [BUG] re-open camera streaming config fails with unexpected code {acq_code} / -16

irimitenkan, 2023-02-19:
**Describe the bug**
Based on examples/capture_stream_udp.py it I can create a stream  successfully , but after stop_recording() it is not possible restart UDP stream again. 
The problem is similar to closed issue #483

Used version picamera 0.3.8

**To Reproduce**
`
import socket
import time

from picamera2 import Picamera2
from picamera2.encoders import H264Encoder
from picamera2.outputs import FileOutput

picam2 = Picamera2()
video_config = picam2.create_video_configuration({"size": (1280, 720)})
picam2.configure(video_config)
encoder = H264Encoder(100000)

with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as sock:
    sock.connect(("192.168.xxx.xxx", 10001))
    stream = sock.makefile("wb")
    picam2.start_recording(encoder, FileOutput(stream))
    time.sleep(20)
    picam2.stop_recording()
    picam2.stop()
    sock.close()

picam2 = Picamera2()
video_config = picam2.create_video_configuration({"size": (1280, 720)})
picam2.configure(video_config)
encoder = H264Encoder(100000)

with socket.socket(socket.AF_INET, socket.SOCK_DGRAM) as sock:
    sock.connect(("192.168.xxx.xxx", 10001))
    stream = sock.makefile("wb")
    picam2.start_recording(encoder, FileOutput(stream))
    time.sleep(20)
    picam2.stop_recording()
    picam2.stop()
    sock.close()

`
**Expected behaviour**
Restarting the UDP stream again when I create a new worker thread in my applicaton

**Console Output, Screenshots**
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 242, in __init__
    self._open_camera()
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 439, in _open_camera
    raise RuntimeError(f"camera.acquire() returned unexpected code: {acq_code}")
RuntimeError: camera.acquire() returned unexpected code: -16

**Hardware :**
RPI 3B+ and RASP CAM 3

irimitenkan, 2023-02-20:
> Hi, I expect the previous Picamera2 instance isn't closed when you open the new one. Perhaps use `picam2.close()` before creating the new instance, or maybe wrap your use of Picamera2 objects in a `with Picamera2() as picam2:`. Does that help?

Hi, thank you . That helps ! I was no aware of the the close() method only the stop(). Hence #576 can be closed.

davidplowman, 2023-02-20:
Thanks for the confirmation!

closed 2023-02-20 by davidplowman