raspberrypi/picamera2 #581: Picamera2 change configuration through switch mode is not working (hangs). 64MP Camera

garg4s, 2023-02-21:
In Picamera2, whenever we change configuration through switch mode it get hangs on capture through any method or any size. 
Without changing any configuration its work perfectly.  We want live stream at 30fps & capture photo on high resolution.
Thanks 
**Hardware :**
 64MP Camera module, PI 4 8GB, 64bit OS

davidplowman, 2023-02-21:
Hi, if you could post a minimal test script that shows the problem then we could comment on whether it looks correct or not. It would be interesting to know if there are any error messages anywhere, perhaps in `dmesg`.

Beyond that my suspicion would be that the problem is related either to the very large amount of memory this sensor requires, or to the sensor driver, in which case you'd have to reach out to the camera vendor for support.

davidplowman, 2023-03-09:
Closing because we believe this problem to be specific to this Arducam camera module. If you experience a similar issue with a Raspberry Pi camera, please do open a new issue. Thanks!

closed 2023-03-09 by davidplowman