raspberrypi/picamera2 #583: Example Thread Class for Pi Camera Array

pageauc, 2023-02-22:
I have written a picamera2 class that runs pi camera capture array in a thread similar to Imutils picamera stream by Adrian Rosebrock.  You may want to include in your Examples.
Class example here
https://raw.githubusercontent.com/pageauc/MoTrack-Picam2-Demo/master/streampilibcam.py

Demo code here  https://github.com/pageauc/MoTrack-Picam2-Demo

davidplowman, 2023-02-22:
Hi, and thank you for the example.

I was wondering if you had any (very simple!) example code that showed how to drive it, and why one might use it. Thanks!

ballen4705, 2023-02-26:
> I have written a picamera2 class that runs pi camera capture array in a thread

Your thread has a loop which sleeps for 10 milliseconds, then calls capture_array.  Another way to get regular access to an in-memory copy of the camera data is to attach a callback function to the picamera2 loop.  See `pre_callback` and `post_callback` in sections 8.2 and 9.4 of _The Picamera2 Library_, which also contains several examples. My understanding is that this avoids the overhead of copying.  To quote the manual: "_Finally we draw attention to the MappedArray class. This class is provided as a convenient way to gain in-place access to the camera buffers - all the capture methods that applications normally use are returning copies._"

davidplowman, 2023-03-13:
Hi, I'm thinking of converting this to a discussion rather than an "issue", would that make sense? Thanks!