raspberrypi/picamera2 #584: [HOW-TO]How to solve camera color deviation

yingliuzhizhuzed, 2023-02-23:
**Please only ask one question per issue!**
In fact, my light source is a green laser light source, and I added a green light filter in front of my camera. Then the problem comes. The color in the image I get is purple, which is not normal for me. I don't know much about the color space of the camera.
**Describe what it is that you want to accomplish**
Now the picture is like this：
[image]
I wish my picture like this（by my phone）
[image]

davidplowman, 2023-02-23:
The AWB algorithm is calibrated for the kinds of lights that we find in typical indoor and outdoor environments. As your light source sounds different, you probably need to set the colour gains manually, for example
```
picam2.set_controls({'ColourGains': (1.5, 1.5)})
```
The first of the two numbers is the red gain, and the second is the blue gain. You would need to adjust them to get the result that you want. Hope that helps!

yingliuzhizhuzed, 2023-02-24:
Thank you，i will try it.In addition, the camera I am using is not the official camera. I am considering whether to use the official V3 camera. Maybe his performance will be better.

closed 2023-02-28 by yingliuzhizhuzed