raspberrypi/picamera2 #586: [BUG] Display DRM Preview on TFT

Kool-Whipp, 2023-02-23:
I'm attempting to display the camera's preview on a TFT using the DRM preview. I run the below python script:

#!/usr/bin/python3

# For use from the login console, when not running X Windows.

import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.DRM, x=0, y=0, width=240, height=240)

preview_config = picam2.create_preview_configuration({"size": (240, 240)})
picam2.configure(preview_config)

picam2.start()
time.sleep(5)

I get the following Error:

[0:10:15.711574226] [820]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[0:10:15.866380010] [821]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media3 and ISP device /dev/media0
[0:10:15.932925487] [820]  INFO Camera camera.cpp:1028 configuring streams: (0) 240x240-XBGR8888
[0:10:15.935368468] [821]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 640x480-SGBRG10_1X10 - Selected unicam format: 640x480-pGAA
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/picamera2/previews/null_preview.py", line 29, in thread_func
    callback(picam2)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 75, in handle_request
    picam2.process_requests(self)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 1136, in process_requests
    display.render_request(display_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 66, in render_request
    self.render_drm(self.picam2, completed_request)
  File "/usr/lib/python3/dist-packages/picamera2/previews/drm_preview.py", line 138, in render_drm
    raise RuntimeError("Failed to reserve DRM plane")
RuntimeError: Failed to reserve DRM plane

It operates fine when an HDMI display is plugged in but not the TFT display.

davidplowman, 2023-02-24:
@6by9 Any advice on this one? I have no idea if/how you could make DRM work with a panel like this...    Thanks!

6by9, 2023-02-24:
What is this TFT display? DPI, DSI, SPI, I2C, or some other interface? What resolution?

DSI, DPI, HDMI, and composite video all go through the Pi's HVS and therefore support numerous source formats (including YUV420). They should all work with Picamera2.

SPI and I2C displays tend to only be RGB only, frequently have no DRM driver, and are very slow to update.
Picamera2 could theoretically produce RGB for the preview stream (I don't know if that selection mechanism is implemented), but it would also need to select the primary DRM plane rather than an overlay plane on these displays.

Kool-Whipp, 2023-02-24:
Display info:
- 240 x 240
- SPI
- Using using an Adafruit Python script to display the console

6by9, 2023-02-24:
Largely not a hope then.

I'm guessing https://www.adafruit.com/product/4484 which is 1.3" 240x240 - there is no DRM driver for it.

There is a DRM driver for the [2.8"](https://www.adafruit.com/product/1601) and [3.5"](https://www.adafruit.com/product/2097) PITFT displays that also connect over SPI, but I've not tried them with Picamera. I suspect that they would also not be able to find a compatible plane for the reasons I'd listed above.

Kool-Whipp, 2023-02-24:
That would be correct. That's a shame then... Thank you for the quick turnaround on the issue!

closed 2023-02-24 by Kool-Whipp