raspberrypi/picamera2 #587: [OTHER]Picamera2 and libcamera Issue

Msmhgn, 2023-02-24:
Hi,

Working with quad-camera imx477 HAT,  when I want to use Picamera2 for multiple capturing I got this error:

@python test.py
[0:02:46.040444550] [1797] INFO Camera camera_manager.cpp:299 libcamera v0.0.0+4186-d1dc0373
[0:02:46.063683715] [1798] WARN RPI raspberrypi.cpp:1308 Mismatch between Unicam and CamHelper for embedded data usage!
[0:02:46.064559799] [1798] INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media3 and ISP device /dev/media0
Traceback (most recent call last):
File “/home/pi/test.py”, line 8, in
“AwbMode”: libcamera.controls.AwbModeEnum.Daylight})
NameError: name ‘libcamera’ is not defined

I tried to install libcamera from [HERE](https://www.arducam.com/downloads/synchronized-quad-camera-kit-getting-started.pdf), then I got this error:

@./install_pivariety_pkgs.sh -p libcamera_dev
=================================================
Hardware Revision: b03111
Kernel Version: 5.10.92-v7l+
OS Codename: bullseye
ARCH: armv7l
kernel:5.10.92-v7l+
–2023-02-23 11:10:18-- https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/libcamera_dev_links.txt
Resolving [github.com](http://github.com/) ([github.com](http://github.com/))… 140.82.113.3
Connecting to [github.com](http://github.com/) ([github.com](http://github.com/))|140.82.113.3|:443… connected.
HTTP request sent, awaiting response… 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/26dffcaf-16dc-499f-9891-18a8b34bcb79?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230223%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230223T161018Z&X-Amz-Expires=300&X-Amz-Signature=0d99bd5aebb576e4a626344e75ddea26a4383c63c61d8a73d57168a6a2e1ee28&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera_dev_links.txt&response-content-type=application%2Foctet-stream [following]
–2023-02-23 11:10:18-- https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/26dffcaf-16dc-499f-9891-18a8b34bcb79?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230223%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230223T161018Z&X-Amz-Expires=300&X-Amz-Signature=0d99bd5aebb576e4a626344e75ddea26a4383c63c61d8a73d57168a6a2e1ee28&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera_dev_links.txt&response-content-type=application%2Foctet-stream
Resolving [objects.githubusercontent.com](http://objects.githubusercontent.com/) ([objects.githubusercontent.com](http://objects.githubusercontent.com/))… 185.199.109.133, 185.199.110.133, 185.199.108.133, …
Connecting to [objects.githubusercontent.com](http://objects.githubusercontent.com/) ([objects.githubusercontent.com](http://objects.githubusercontent.com/))|185.199.109.133|:443… connected.
HTTP request sent, awaiting response… 200 OK
Length: 4095 (4.0K) [application/octet-stream]
Saving to: ‘libcamera_dev_links.txt’

libcamera_dev_links 100%[===================>] 4.00K --.-KB/s in 0.003s

2023-02-23 11:10:18 (1.16 MB/s) - ‘libcamera_dev_links.txt’ saved [4095/4095]

–2023-02-23 11:10:18-- https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/libcamera-v0.0.5/libcamera-dev-0.0.12-bullseye-armhf.deb
Resolving [github.com](http://github.com/) ([github.com](http://github.com/))… 140.82.113.3
Connecting to [github.com](http://github.com/) ([github.com](http://github.com/))|140.82.113.3|:443… connected.
HTTP request sent, awaiting response… 302 Found
Location: https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/1496d493-b770-4703-9db7-bbfd8655f59a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230223%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230223T161018Z&X-Amz-Expires=300&X-Amz-Signature=e31c9452f46e8b69c40b3f45ab5d733395a357ad4987f640138fac81860749bf&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera-dev-0.0.12-bullseye-armhf.deb&response-content-type=application%2Foctet-stream [following]
–2023-02-23 11:10:18-- https://objects.githubusercontent.com/github-production-release-asset-2e65be/353945933/1496d493-b770-4703-9db7-bbfd8655f59a?X-Amz-Algorithm=AWS4-HMAC-SHA256&X-Amz-Credential=AKIAIWNJYAX4CSVEH53A%2F20230223%2Fus-east-1%2Fs3%2Faws4_request&X-Amz-Date=20230223T161018Z&X-Amz-Expires=300&X-Amz-Signature=e31c9452f46e8b69c40b3f45ab5d733395a357ad4987f640138fac81860749bf&X-Amz-SignedHeaders=host&actor_id=0&key_id=0&repo_id=353945933&response-content-disposition=attachment%3B%20filename%3Dlibcamera-dev-0.0.12-bullseye-armhf.deb&response-content-type=application%2Foctet-stream
Resolving [objects.githubusercontent.com](http://objects.githubusercontent.com/) ([objects.githubusercontent.com](http://objects.githubusercontent.com/))… 185.199.111.133, 185.199.109.133, 185.199.110.133, …
Connecting to [objects.githubusercontent.com](http://objects.githubusercontent.com/) ([objects.githubusercontent.com](http://objects.githubusercontent.com/))|185.199.111.133|:443… connected.
HTTP request sent, awaiting response… 200 OK
Length: 964530 (942K)

Unknown error, please send the error message to [support@arducam.com](mailto:support@arducam.com)

Do you have any idea what is the issue? what process should I follow to make sure no issues will happen?

Thanks

davidplowman, 2023-02-24:
Thanks for the report. It's a bit hard to guess what's happened there. It seems to be unable to open the camera, possibly because something else has ended up in a state where it's holding on to it and hasn't "died" properly. I presume you've tried a reboot to see if that clears the problem?

Obviously if you can pin down what you did to make it get into this state then we'd obviously want to look into it. Thanks.

Msmhgn, 2023-03-06:
Thanks @davidplowman 
I reinstalled the picamera2 and RPI and the issue solved.

closed 2023-03-06 by Msmhgn