raspberrypi/picamera2 #590: [HOW-TO] Timlapse with .DNG and .JPG while keeping Preview between captures

TA3ASC0, 2023-02-26:
Hello everyone,

it's my first post here so I'm happy about every advise if I may have missed a rule.

I want to create a timelapse script that captures both, DNG and JPG files over a definied period of time.
Between the captures, I want to see the QTGL preview window.

At the moment I'm stuck in an early state of prototyping.
I used the pdf manual and the examples of 

- [capturing dng and jpg with a request](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_dng_and_jpeg.py)
- [capturing dng and jpg with helpers](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_dng_and_jpeg_helpers.py)
- [capturing a timelapse](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_timelapse.py)

I manged to create a timelapse with jpg and dng _without_ preview (modifying the example code).
Here is my attempt that worked so far. It's a modified version of the timelapse example.
The issue is that I can't see the preview once I start capturing.

```
#!/usr/bin/python3
import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

capture_config = picam2.create_still_configuration(raw={}, display=None)
preview_config = picam2.create_preview_configuration()

picam2.configure(preview_config)
picam2.start()

# Give time for Aec and Awb to settle, before disabling them
time.sleep(1)
picam2.set_controls({"AeEnable": False, "AwbEnable": False, "FrameRate": 1.0})
# And wait for those settings to take effect
time.sleep(5)

picam2.switch_mode(capture_config)

start_time = time.time()
for i in range(1, 4):
    r = picam2.capture_request()
    r.save("main", f"image{i}.jpg")
    r.save_dng(f"image{i}.dng")
    r.release()
    print(f"Captured image {i} of 50 at {time.time() - start_time:.2f}s")
    time.sleep(2)

picam2.stop()
```
I modified this code to stop, reload the preview config and start picam2 once a picture is taken:
```
#!/usr/bin/python3
import time

from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.QTGL)

capture_config = picam2.create_still_configuration(raw={}, display=None)
preview_config = picam2.create_preview_configuration()

picam2.configure(preview_config)
picam2.start()

# Give time for Aec and Awb to settle, before disabling them
time.sleep(1)
picam2.set_controls({"AeEnable": False, "AwbEnable": False, "FrameRate": 1.0})
# And wait for those settings to take effect
time.sleep(5)

start_time = time.time()
for i in range(1, 4):
    picam2.stop()
    picam2.configure(capture_config)
    picam2.start()
    r = picam2.capture_request()
    r.save("main", f"image{i}.jpg")
    r.save_dng(f"image{i}.dng")
    r.release()
    print(f"Captured image {i} of 50 at {time.time() - start_time:.2f}s")
    picam2.stop()
    picam2.configure(preview_config)
    picam2.start()
    time.sleep(4)

picam2.stop()
```

Doing so, the preview works again after the first image is taken.
But as soon as I want to switch to the still configuration for the _second image_, the script crashes:

```
[1:49:42.994909708] [4492]  INFO Camera camera_manager.cpp:299 libcamera v0.0.3+40-9b860a66
[1:49:43.054189571] [4495]  INFO RPI raspberrypi.cpp:1425 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media2
[1:49:46.387241577] [4492]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-XBGR8888
[1:49:46.387866359] [4495]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[1:49:53.539985187] [4499]  INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-BGR888 (1) 4056x3040-SBGGR12_CSI2P
[1:49:53.543667920] [4495]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Captured image 1 of 50 at 3.96s
[1:49:57.369855461] [4499]  INFO Camera camera.cpp:1028 configuring streams: (0) 640x480-XBGR8888
[1:49:57.389531711] [4495]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC
[1:50:02.081533181] [4499]  INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-BGR888 (1) 4056x3040-SBGGR12_CSI2P
[1:50:02.084077125] [4495]  INFO RPI raspberrypi.cpp:805 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pBCC
Traceback (most recent call last):
  File "/home/pi/Documents/Code/PBCam/tests/github_timelapse_04.py", line 24, in <module>
    picam2.configure(capture_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 975, in configure
    self.configure_(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/picamera2.py", line 968, in configure_
    self.video_configuration.update(camera_config)
  File "/usr/lib/python3/dist-packages/picamera2/configuration.py", line 59, in update
    self.__setattr__(k, v)
  File "/usr/lib/python3/dist-packages/picamera2/configuration.py", line 43, in setattr
    value = self._FIELD_CLASS_MAP[name](value)
  File "/usr/lib/python3/dist-packages/picamera2/configuration.py", line 35, in init
    self.__setattr__(k, v)
  File "/usr/lib/python3/dist-packages/picamera2/configuration.py", line 46, in setattr
    raise RuntimeError(f"Invalid field '{name}'")
RuntimeError: Invalid field 'bpp'
```

I found a similar issue with the same error that has been closed [here](https://github.com/raspberrypi/picamera2/issues/318).
But the mentioned ``release()`` is already part of my code.

My attempts to modify the two examples for capturing dng and jpgs always led to the same issue with the ``RuntimeError: Invalid field 'bpp'``
I don't include these examples here since I don't want to overload my post. If they are helpful to solve the issue, I can post them too.

I feel like I'm doing a mistake with buffers and / or creating the requests and maybe I'm doing a very obvious mistake or I may have chosen a fundamental wrong way to create such a task.

Thus, I'm happy about any help rearding this issue.

Thank you very much!

davidplowman, 2023-02-27:
Thanks for reporting this. Yes I can see what's happening. I'll make sure that the fix (https://github.com/raspberrypi/picamera2/pull/593) gets into the next release (which I'm hoping will be very soon).

Petros626, 2023-03-07:
@TA3ASC0 could you provide some hints to create a picture taker with picamera2, when pressing a key?