raspberrypi/picamera2 #594: [HOW-TO] interpreting raw Bayer arrays for the v3 raspberry pi camera module

ajarang, 2023-02-27:
**Describe what it is that you want to accomplish**
I want to figure out how to interpret data from the bayer sensor for the v3 camera module. 

**Describe alternatives you've considered**
I am trying to get and interpret the Bayer data the pi camera module v3. I have exported an array using 
`picam2.caputre_array("raw")`, but the output array seems to repeat every 5 pixels lengthwise instead of 4, as I would expect from an RGGB bayer patterned sensor. I am doing this because I want to isolate the individual channels for each sensor, so that I have individual arrays for each R, G, G, and B "bayer pixel". 

The size of the pixelarray is 4608,2592, but the size of the raw output file is 5760,2592. 5760 isn't evenly divided by 4608. There is some info on the [picamera1 readthedocs](https://picamera.readthedocs.io/en/release-1.13/recipes2.html?highlight=bayer#raw-bayer-data-captures) that demonstrates the conversion procedure between the two, but I'm not sure how these change for the camera module v3.

davidplowman, 2023-02-27:
By default you're getting the packed output format, so each pixel has 10 bits, but 4 pixels are packed into 5 bytes. In this format, the first 4 bytes are the top eight bits of each pixel, and the bottom two bits of each pixel are packed into the 5th bytes. You can see that 4608 * 10 / 8 = 5760, which accounts for the stride.

You may find it easier to ask for _unpacked_ raw pixels. In this case each pixel would occupy 16 bits, and the top 6 bits of each 16-bit word would be zero. If you want to read the raw image using software then this form is often more convenient, and the stride should show 9216, or 2 bytes for each pixel. To obtain unpacked raw pixels, use something like
```
config = picam2.create_still_configuration(raw={'format': 'SBGGR10'})
```
Hope that helps!

ajarang, 2023-02-28:
Thank you for your quick response! Out of curiosity, why is the Packed output format selected by default? Is it for size/compatibility?

Also, I realized that the size of the output from capture_array is 5760 x 2592. However, the maximum resolution of the raspberry pi camera v3 sensor is 4056 x 3040, so I would expect the bayer array size to be 4 times that. Would you happen to know why this could be the case? I would expect "raw" to give me exactly everything from the bayer sensor.

tvoverbeek, 2023-02-28:
The array output by capture_array is a numpy array.
Each array item can have several components, i.e it is a kind of 3d array

ajarang, 2023-02-28:
`raw=picam2.capture_array("raw")` is used to collect what I assume is Raw Bayer data. then I use `savemat("output.mat",{'bayeraw':raw})` to export the array to a .mat file. even running `raw.shape` gives me the size above, which is still unexpected.

tvoverbeek, 2023-02-28:
Max resolution 4056 x 3040 is for the HQ camera, not the V3 camera.
Max resolution for V3 is 4608 x 2592.
See https://www.raspberrypi.com/documentation/accessories/camera.html#hardware-specification

ajarang, 2023-02-28:
sure, but I would expect my bayer array to be 4608*4 (18432) by 2592*4 (10368), since there are 4 filters for each pixel.

tvoverbeek, 2023-02-28:
No, there is only one color filter per pixel. Google for Bayer Pattern or look in Wikipedia

closed 2023-02-28 by ajarang

davidplowman, 2023-03-01:
Just to add a couple of things.

The reason you get packed formats by default is because they're more efficient. Less system memory space (the total amount available is quite limited because it's a "special kind" of memory) and bandwidth being used up. This gets presented to you as 8-bit bytes and you have to unpack the bits correctly.

If you ask for unpacked values you still get bytes, but you can convert them to 16-bit words "for free" using numpy's `view` method.

Note that the Bayer array has only one filter for each pixel. So for the HQ cam, you get 4056x3040 numbers, where each pixel gives you only one colour, not three. All the missing colours are interpolated. Hope that makes sense!