raspberrypi/picamera2 #602: [SUGGESTION] Saving of LibRaw compatible RAW image (array & meta data)

patrickwolf, 2023-03-04:
**Is your feature request related to a problem? Please describe.**
PiCam2 provides multiple streams: RAW/DNG/JPG
They all have different file sizes and take different processing times to create
ie creating a 64MP RAW and saving it to disk is 80%+ faster then converting the RAW to DNG and then saving that

So saving out the RAW directly for post processing in other applications can be very efficient!

**Describe the solution you'd like**
I would like to have a helper function that can save RAW (data array + meta data) into a file that is readable by RAW libraries like https://www.libraw.org (and with that 3rd party RAW compatible programs)

**Describe alternatives you've considered**
Saving RAW as two files img.array & img.meta and then load them in again later to convert to DNG on the RPI

**Additional context**
A related ticket that goes into saving raws #311

Bra1nsen, 2023-03-04:
you can save the raw files and the metadata information in a numpy array of your choice.

Bra1nsen, 2023-03-04:
whats your use case?

patrickwolf, 2023-03-04:
My use case is achieving a high FPS for capturing 250+ 64MP raw images to disk.
Converting to DNG takes 90% longer hence I'm needing RAWs.

If the RAWs were in the right format (readable by other programs) the converting to DNG could be scaled and done in post after the capture session (ie wouldn't have to happen on the RPI).

Bra1nsen, 2023-03-04:
which raw format is needed for your 3rd party conversion programm?

what for do you need high fps? is it a real time monitoring application?

why do you need dng?

patrickwolf, 2023-03-05:
> which raw format is needed for your 3rd party conversion program?

any raw format https://www.libraw.org/ could open

> what for do you need high fps? is it a real time monitoring application?

to shoot a scene and capture as much detail and as many pictures as possible ie 64MP & raw & high FPS

> why do you need dng?

If the RAWs were in the right format (readable by other programs) then they could be used directly in 3rd party programs.. so DNGs are needed because I can't read the raw format that is being created right now

What I'm really looking for is an answer to my original question: 
- We need high resolution and high FPS. 
- Saving DNGs is too slow
- Saving RAWs arrays are fast enough but they are not readable by 3rd party programs
- Is there anyway we can store the RAW (data & metadata) in a format that is compatible with one of the https://www.libraw.org/ supported formats?

patrickwolf, 2023-03-05:
So I asked ChatGPT :) 

Given a numpy array with camera data and this metadata:
```
{"SensorTimestamp": 93634696093000, "ScalerCrop": [0, 40, 9152, 6864], "FocusFoM": 54915287, "Lux": 4755910.0, "FrameDuration": 370366, "AnalogueGain": 1.0, "AfPauseState": 0, "ColourCorrectionMatrix": [1.5325424671173096, -0.2595469355583191, -0.2729856073856354, -0.32578086853027344, 1.574324131011963, -0.24854332208633423, -0.1713532656431198, -0.30753612518310547, 1.4788893461227417], "AeLocked": false, "SensorBlackLevels": [4096, 4096, 4096, 4096], "ColourGains": [2.1177597045898438, 1.596767544746399], "DigitalGain": 1.0, "ExposureTime": 9977, "ColourTemperature": 6492, "AfState": 0}
{"format": "SBGGR10_CSI2P", "size": [9152, 6944], "stride": 11456, "framesize": 79550464}
```
How can i save this in python as raw image that is recognized by libraw? 

I'll save the answer in a new message and would love community support in evaluating its correctness

patrickwolf, 2023-03-05:
ChatGPT

> To create a raw image from the sensor data and metadata, you will need to convert the data into a format that can be written to a file as a raw image. One common format for raw image data is Bayer pattern, which represents each pixel as a single color channel value.
> 
> Here's an example of how you can convert the sensor data and metadata into a raw Bayer pattern image in Python:

```
import numpy as np
import struct

# Load the sensor data from a numpy array
data = np.load('sensor_data.npy')

# Load the metadata from a JSON file
import json
with open('metadata.json', 'r') as f:
    metadata = json.load(f)

# Extract the relevant metadata values
size = metadata['size']
format = metadata['format']
crop = metadata['ScalerCrop']

# Extract the red, green, and blue color channels from the sensor data
r = data[0::2, 0::2]
g1 = data[0::2, 1::2]
g2 = data[1::2, 0::2]
b = data[1::2, 1::2]

# Combine the color channels into a Bayer pattern image
bayer = np.zeros((size[1], size[0]), dtype=np.uint16)
bayer[0::2, 0::2] = r
bayer[0::2, 1::2] = g1
bayer[1::2, 0::2] = g2
bayer[1::2, 1::2] = b

# Apply the scaler crop to the Bayer pattern image
bayer = bayer[crop[1]:crop[3], crop[0]:crop[2]]

# Write the raw image data and metadata to a file
with open('image.raw', 'wb') as f:
    # Write the raw image data as a byte string
    f.write(bayer.tobytes())

    # Write the metadata as a JSON string
    metadata_str = json.dumps(metadata)
    f.write(struct.pack('I', len(metadata_str)))
    f.write(metadata_str.encode())

```

> In this example, we first load the sensor data from a numpy array and the metadata from a JSON file. We then extract the relevant metadata values, including the image size, format, and crop region.
> 
> Next, we extract the red, green, and blue color channels from the sensor data and combine them into a Bayer pattern image. We apply the scaler crop to the Bayer pattern image and write the raw image data as a byte string to a file.
> 
> Finally, we encode the metadata as a JSON string and write it to the same file, prefixed by a four-byte integer indicating the length of the metadata string. This allows third-party applications to read the raw image data and metadata together. Note that the format of the metadata may need to be adjusted depending on the requirements of the application that will be reading the raw image data.

I'm missing the raw data experience to evaluate the statements and would appreciate communicate support. 

Is this a viable way to store my numpy array and metadata from my cam sensor as a raw file that is recognized?

davidplowman, 2023-03-06:
The question of saving raw files quickly comes up from time to time, and each time it does the discussion goes in a slightly different direction. It came up on the camera board forum again recently: https://forums.raspberrypi.com/viewtopic.php?t=347717#p2083384

On that occasion, the solution adopted was simply to hold the raw image buffers in memory (it was an 8GB device and could therefore store a very large number), and save them as DNGs after (so far as I know, DNG is the only readily portable format). Might that work for you?

With a 64MP sensor, of course, that will be considerably more challenging, and the achievable framerates will be quite low.

Bra1nsen, 2023-03-06:
> to shoot a scene and capture as much detail and as many pictures as possible ie 64MP & raw & high FPS

hat sounds very interesting, can you give some more context about the following image processing pipeline?
--> Its pretty tough tho, to handle so much raw-data, which scene are you capturing that you need 64 MP & high fps?

> Is this a viable way to store my numpy array and metadata from my cam sensor as a raw file that is recognized?

yes that looks legit, you could still think about cloud computing so that the Pi just uploads the raw files and the further processing is done with more computational power. Which camera module is connected (Arducam?), Libraw got a long list of supported cams, so I guess there are multiple raw file formats possible..

davidplowman, 2023-03-13:
Hi, just wondering if there is anything else for us to investigate here? Thanks!

patrickwolf, 2023-03-14:
>  the Pi just uploads the raw files and the further processing is done with more computational power

@Bra1nsen I appreciate your effort to see how the pipeline could be improved and yet my question is really just around how to allow PiCam2 to store RAW files directly instead of needing to go via DNGs (the pipeline is more complex then what i described here) 

> yes that looks legit (@Bra1nsen )

Great thanks

patrickwolf, 2023-03-14:
> Hi, just wondering if there is anything else for us to investigate here? Thanks!

considering you are a major contributor to PiCam2 @davidplowman could you let me know if there is 
- merit to the ChatGPT solution above for saving LibRaw compatible Raw files 
- if there is interest to add saving to Raw (without DNG processing) as a helper function to PiCam2?

Thanks,
Patrick

davidplowman, 2023-03-15:
Hi, I guess the first thing is to be sure I'm understanding your requirements properly.

I believe that you want to be able to save raw image files in a generally recognised format, but using Picamera2 to save DNGs is too slow. Instead you want to find another format that is quicker to save.

So you need to find a (fairly) widely supported raw file format that is quick to save. As I said earlier, I don't know of one. DNG is widely supported, but slow. Perhaps it could be speeded up? You'd have to figure out what it does and whether there's room for improvement.

There are also many proprietary camera manufacturer formats, but how easy it would be to save them, I wouldn't like to say. Maybe you can find Python modules that would write them, but I'm not very optimistic.

As regards the ChatGPT code, it seems to be reading a big numpy array (the raw data), and some metadata. Presumably the idea is that you would have saved these earlier, which is fair enough. Then it copies the raw image from `data` into `bayer`. As far as I can see this is just a copy, and is pointless. Then it crops the array according to `'ScalerCrop'`, which is not normally something we do to the raw camera data - the idea is to save everything that the camera gave you, not to throw away some part of it if the user has applied some digital zoom.

Finally it merely saves the array and the metadata into a single file - which you might as well have done in the first place, as far as I understand it. So unless I've misunderstood what you want, or what it's doing, it doesn't seem helpful to me.

I would have thought the most likely options for you to investigate would be either to make saving DNGs faster, or to use your own file format that you can write quickly - you could always convert them to DNG later for interoperability.

davidplowman, 2023-04-04:
Hi, is there anything else for us to investigate here? Thanks.

davidplowman, 2023-04-11:
I'll close this issue for now, if that's ok, but please do open another one if there are still further things to look at. Thanks!

closed 2023-04-11 by davidplowman