raspberrypi/picamera2 #604: Exposure time change

Msmhgn, 2023-03-06:
Hi all,
Working with Picamera2, I wanted to capture several images with different exposure/shutter speeds. For me, big numbers for exposure time do not work at all. At some point, It apparently gives me the same image. I worked with the exposure_fixed example and changed the number, but no change in the image for big numbers like 1000000, and 20000000.
I am wondering what could be the issue.

Would appreciate it if anyone could help me.

tvoverbeek, 2023-03-06:
Which camera are you using?
For the raspberry pi cameras see this table: https://www.raspberrypi.com/documentation/accessories/camera.html#hardware-specification to find the max exposure time
The V1 exposure time for example is 6 seconds.

davidplowman, 2023-03-09:
When you have captured the files, do the exif tags report the exposure that you are expecting?

thestfndev, 2023-03-09:
I have a similar issue, I want to set up picamera2 to make totally manual long exposures, but anything I try, I cannot make them longer than 1.7 seconds.
`camera_controls()` returns

`{...'ExposureTime': (26, 1722331, None)...}`

Camera Module 3 on RPi 4B with Raspbian.

davidplowman, 2023-04-11:
I'll close this one for now, if that's ok, but please do open another issue if there are still problems. Thanks!

closed 2023-04-11 by davidplowman