raspberrypi/picamera2 #608: Documentation of a function

Petros626, 2023-03-08:
Hey folks,

I dig into the code in `picamera2.py` to see, what arguments I can pass to `capture_file()`. From source code it seems that, you can 
pass these arguments without any problem. But in my case there is no image, which is saved.

```python
def capture_file(
            self,
            file_output,
            name: str = "main",
            format=None,
            wait=None,
            signal_function=None) -> dict:
```
I used this function for my purpose like this:
```python
filename =  time.strftime("%Y%m%d-%H%M%S") + '.png'
output = '/home/pk/Pictures'
cam.capture_file(output, filename, wait=False)

# OR

filename = dirname + "_" + str(imgnum) + '.png'
savepath = os.path.join(dirpath, filename)
cam.capture_file(savepath, format='png', wait=False)
```

Why nothing of this cases is working? The only way it's shown in examples is using `capture_file(path/to/image/test.png)`

davidplowman, 2023-03-08:
Hi, your first example doesn't work because the first (required) parameter must be a file name, and the second would be the name of the stream ("main", "lores" or "raw", defaulting to "main").

The second example works for me - can you say how it is failing?

Petros626, 2023-03-08:
Why it's not like the handled arguments in the definition of this function? 

With your changed suggestions it prints, that the picture is taken, but there's **no picture taken** and saved (default location: maybe `/home/username`). The same with the second example, which works for you.
```python
cam.capture_file(filename, "main", format='png', wait=False)
cam.capture_file(savepath, format='png', wait=False)
```
The second examples don't contain the filename, just the save path, which includes the filename. So did you change anyhing else? How can I o. it's possible to set the save location?

To prevent misunderstandings I'm using this:
```python
def getkey():
    fd = sys.stdin.fileno()
    old_settings = termios.tcgetattr(fd)
    try:
        tty.setraw(fd)
        ch = sys.stdin.read(1)
    finally:
        termios.tcsetattr(fd, termios.TCSADRAIN, old_settings)
    return ch

dirname = 'home/pk/Pictures/'
cwd = os.getcwd()
dirpath = os.path.join(cwd,dirname)

cam = Picamera2()
time.sleep(1)

try:
    while True:
        key = getkey()
        if key == 'p':
            filename =  time.strftime("%Y%m%d-%H%M%S") + '.png'
            savepath = os.path.join(dirpath, filename)
            cam.capture_file(savepath, format='png', wait=False)
            #cam.capture_file(filename, "main", format='png', wait=False)
            print(f"Captured {filename} succesfully")
        elif key == 'q':
            print("Closing camera...")
            break
finally:
    cam.close()
```

davidplowman, 2023-03-08:
Hi, could you maybe try the following and report whether it works:
```
from picamera2 import Picamera2
cam = Picamera2()
cam.start()
cam.capture_file("test.png", format='png')
```
(As I said somewhere else, I wouldn't use `wait=False`.)

I'm afraid I couldn't quite understand exactly what else was not working as you expected. If you could include a minimal test case (as I have above), report exactly how it behaves incorrectly, along with any error messages, that would be very helpful. Thank you!

Petros626, 2023-03-09:
Hey,
it works fine, can you **link** the issue where you spoke about the use of `wait=False`. 

My intention was to save my file to a specific folder with a specific filename, the **WRONG** listing of handling parameters in the fucntion `cam.capture_file` has prevented that without any warning or error message. Was just a pure logic error.

I would like to share my example script through a pull request, so other don't struggle to find exmaples like mine.

davidplowman, 2023-03-09:
> Hey, it works fine, can you **link** the issue where you spoke about the use of `wait=False`.

I mentioned it [here](https://github.com/raspberrypi/picamera2/issues/606#issuecomment-1458402671).

> 
> My intention was to save my file to a specific folder with a specific filename, the **WRONG** listing of handling parameters in the fucntion `cam.capture_file` has prevented that without any warning or error message. Was just a pure logic error.

I'm afraid I still haven't really understood what was handled incorrectly or did not behave helpfully. Or is it that some documentation is incorrect? If you could include a tiny code snippet and just explain what happened vs what you expected then I would take a look. Thanks!

> 
> I would like to share my example script through a pull request, so other don't struggle to find exmaples like mine.

Yes, please do. Please check the [contribution guidelines](https://github.com/raspberrypi/picamera2#contributing) first.  I don't think there's anything too difficult there (especially if you're only adding an example), but we do have some small conventions about commit messages, sign-off, and which branch to make pull requests for. Thanks!

Petros626, 2023-03-09:
Thanks, thought it was a different issue not (https://github.com/raspberrypi/picamera2/issues/606#issuecomment-1458402671). I will change my current pull request then to your guidelines.

Back to the previous issue:

false code:
https://github.com/raspberrypi/picamera2/issues/608#issue-1615017474
expected behaviour: saving a .png image to the specified folder
happened: not saving anything, just printed in terminal, that a image has been saved

right code:
```python
filename = "".join([dirname, "_", str(imgnum), ".png"]) 
savepath = os.path.join(dirpath, filename)
cam.capture_file(savepath,  "main", format="png", wait=None)
```

expected behaviour: saving a .png image to the specified folder
happened: with your hints here (https://github.com/raspberrypi/picamera2/issues/608#issuecomment-1459958263) it works.

tvoverbeek, 2023-03-09:
In your code dirname has to be an absolute path, i.e. `/home/pk/Pictures` not `home/pk/Pictures` which depending on what is your current directory probably does not exist (unless it is `/`).