raspberrypi/picamera2 #609: [HOW-TO] flickering lights / Hz change?

RolandasRazma, 2023-03-08:
Some of my videos are filtering since migrating to raspberry pi camera v3 and picamera2 when there is an external light bulb. On USB cameras there is usually power line frequency change to workaround that. Is there anything like that on picamera2?

davidplowman, 2023-03-08:
There was some discussion of this on the libcamera mailing list a little while ago, but there haven't been any updates.  The AGC/AEC on the Pi is able to avoid 50/60Hz flicker by using multiples of the lighting period, but there is no API to enable it. Sorry not to have better news on that.

RolandasRazma, 2023-03-10:
I see, thanks for the answer.

closed 2023-03-10 by RolandasRazma

RolandasRazma, 2023-07-26:
looks like it was implemented in libcamera-apps https://github.com/raspberrypi//pull/544/files

davidplowman, 2023-07-26:
Yes, once we pick up an updated version of libcamera these controls will just appear automagically in picamera2 as camera controls.