raspberrypi/picamera2 #612: AttributeError: 'NoneType' object has no attribute 'get' in request.py

Petros626, 2023-03-09:
Following short example leads me to this error:
```python
.
.
.
# imports and def getkey() above...

cam = Picamera2()
cam.start_preview(Preview.QTGL)
cam.start()
request = cam.capture_request()
savepath = "/home/pk/camera/image/"
imgname = "picture"

try:
    while 1:
        key = getkey()
        if key == 'p':
            request.save("main", savepath + imgname, format="png")
            print("Picture taken and saved")
            request.release() # <-- problem seems here
        elif key == 'q':
            print("Closing camera...")
            break
finally:
    cam.stop_preview()
    cam.stop()
    cam.close()
```
After executing the script I get the following message, but with paste `request.release` for example before `finally:`, it seems not to be a problem.
[image]

davidplowman, 2023-03-09:
Hi, it may just be that you need to move the line `request = cam.capture_request()` to just before `request.save(...)`. You need to capture a request every time you want a new image. Once you release a request, you can't use it again. Does that explain it?

Petros626, 2023-03-09:
I understand, thanks.

closed 2023-03-09 by Petros626