raspberrypi/picamera2 #616: Speed up execution with Picamera2

Petros626, 2023-03-11:
Hey,

I've seen that the library `PiCamera2` is using the `threading module`, could it be possible, if you use for example this module in your python script (imager taker or video recorder) to get additional execution speed or does the library already offer the maximum of speed? 

It's just a question of interest, thanks.

davidplowman, 2023-03-13:
Thanks for the question. You'd probably have to be a bit more specific about what you want to achieve. Picamera2 will give you frames pretty much as fast as the camera will deliver them, obviously it's up to the application how it then processes them. Some operations may benefit from threading within numpy or other libraries, or maybe an application could implement multi-threaded algorithms for itself.

The h.264 and hardware MJPEG encoders use built-in hardware, so there would be no benefit from threading there. The software JpegEncoder, when used for MJPEG, uses 4 cores already, so it's not clear to me that there are big and easy improvements available for video recording.

davidplowman, 2023-03-13:
By default many operations actually run in the camera thread (`cam.capture_file()` for example), though the calling thread will normally want to block for it to finish. You can choose not to block for it to finish (`wait=False`), but at some point you will need to block to be sue the file has been written. Maybe this is what you want to do?

Delegating these operations to the camera thread is often convenient but sometimes users prefer not to make the camera thread do extra work. In this case they prefer to use their own thread to do the work. The easiest solution here is often to capture the request and then call `save` in your own thread.

Obviously you're also free to create another thread or task which will perform this operation outside your own thread. At this point, I think it's probably up to the application to implement the behaviour that it needs.

davidplowman, 2023-04-04:
Hi, is there anything else to investigate here or should we close this issue? Thanks.

closed 2023-04-04 by Petros626