raspberrypi/picamera2 #617: [HOW-TO] OpenCV VideoCapture help

Jeevaakg, 2023-03-11:
How to capture pi camera feed to opencv?

I want to us OpenCV.videocaputure () function in my code. Please help me to configure picamera2 for my requirement

Petros626, 2023-03-11:
Hey,

I opened an issue some days before, exactly about this topic here: https://github.com/opencv/opencv/issues/23328. It's not impossible, but you need a little workaround. I'm working on it these days and if I got a working solution, I'll post it here.

davidplowman, 2023-03-13:
You would have to talk to the OpenCV developers about adding support for libcamera, there is not much we can do about this. I believe it was raised at some point, but I don't believe the discussion has really moved on.

As has been commented, the legacy camera stack should work, but you don't get access to the new camera modules, the ability to receive different streams, and so on.

I think some folks have had some luck with the `libcamerify` utility, but it's not an ideal solution.

Sorry not to be more help.