raspberrypi/picamera2 #618: question to raspberry pi camera-module 3

Arnold1, 2023-03-12:
Hi,

I have a few questions to the raspberry pi camera-module 3 and its picamera2 lib:
- can you use the raspberry pi camera-module 3 with Raspberry Pi Pico, Raspberry Pi Zero W, and Raspberry Pi Zero 2 W?
- is the picamera2 lib available for Raspberry Pi Pico, Raspberry Pi Zero W, and Raspberry Pi Zero 2 W?
- when will picamera2 be fully released?

Thanks

davidplowman, 2023-03-13:
> Hi,
> 
> I have a few questions to the raspberry pi camera-module 3 and its picamera2 lib:
> 
> * can you use the raspberry pi camera-module 3 with Raspberry Pi Pico, Raspberry Pi Zero W, and Raspberry Pi Zero 2 W?

You can't use it with the Pico. The Pico has no camera support. You can use it with the Zero W and Zero 2 W (Zero 2 W will perform significantly better).

> * is the picamera2 lib available for Raspberry Pi Pico, Raspberry Pi Zero W, and Raspberry Pi Zero 2 W?
> * when will picamera2 be fully released?

As above, it's not available for the Pico. I'm not quite sure when we shall call it "version 1.0". libcamera is still in development but it would probably make sense to label a full release in the fairly near future once we find we're no longer changing very much.

> 
> Thanks

Arnold1, 2023-03-15:
@davidplowman thanks for the info. Can you do create a webocket connection and stream the video to a server? and the server will provide the live video to other users...

davidplowman, 2023-03-15:
Hi, I don't specifically know anything about websockets, but I'm sure this kind of thing is possible. There are a number of [examples that involve web streaming](https://github.com/raspberrypi/picamera2/tree/main/examples), perhaps these would be helpful?

The `start_recording` method doesn't give you a stream, but if you look at some of those examples you'll see that you can write directly to a socket ([here](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_stream.py)), or create your own output class to do whatever you want ([here](https://github.com/raspberrypi/picamera2/blob/main/examples/mjpeg_server_2.py)).

Arnold1, 2023-03-15:
thank you.

Is there a way to wakeup the raspberry pi zero 2 w and the pi camera module 3 stream the video for 1 minute and go to sleep again - so it saves energy?

davidplowman, 2023-03-15:
I guess it depends what you mean by "sleep". Do you mean just stop running the camera? Or some kind of power saving mode? I'm afraid that's not something I know about - maybe try asking on the regular forums where you might pick up a more knowledge person!

davidplowman, 2023-03-17:
> i mean to power off the camera to save energy and power on again, instead of having it run 24/7.

Probably best to ask on one of the regular forums there, that's not something where I would necessarily know the best answers.

> 
> some other question to the picamera2 lib: i saw it uses the libcamera library which seems to be written in c/c++. what additional features does picamera2 have compared to libcamera? could i also use libcamera lib: https://github.com/kbingham/libcamera and access the raspberry pi camera-module 3 using raspberry pi zero 2?

That repo looks to be a mirror/fork of the [official libcamera repo](https://git.linuxtv.org/libcamera.git/) (kbingham being one of the libcamera developers, of course). Normally we would recommend Pi users to build from our own fork. We do try to keep this exactly in line with the official libcamera, but from time to time "stuff" happens (as libcamera is still in very active development and we need to stay at the bleeding edge) so you're generally safer on the [Raspberry Pi version](https://github.com/raspberrypi/libcamera), where we have verified that things work correctly.

Picamera2 is built entirely on libcamera. Obviously it's Python and most folks would find it much easier to use, but you can program directly against the libcamera C++ API if you prefer. You can compare the experience of writing lower level [C++ applications](https://github.com/raspberrypi/libcamera-apps) against [Python ones](https://github.com/raspberrypi/picamera2/tree/main/examples).

Arnold1, 2023-04-05:
hi, i try to allocate some time this month to look at things.