raspberrypi/picamera2 #619: standard colour channel when saving

Petros626, 2023-03-13:
This function saves .jpg or . png depending on `format=` param. I'm using 'png' and thought it's saved as RGB image, but it's saved as RGBA. Can I change and how to change this behaviour? 
Could it be possible, that using `start_preview(Preview.QTGL)` takes a role here?

https://github.com/raspberrypi/picamera2/blob/d07a49197622d636bd993a5bf1de321c259abf47/picamera2/request.py#L242

Thanks in advance

davidplowman, 2023-03-13:
I remember that there were some strange issues dealing with PIL (Python Image Library) and what it did or did not support. I found that what libcamera calls `XBGR8888` it will only handle if we claim that it's what PIL knows as `RGBA`, so this format gets saved with an alpha channel.

But if you have `XRGB8888` it will only handle that if you call it (in PIL terms) `BGRX`. Presumably PIL decides it can't save this as a PNG and converts it to 24-bit RGB. Basically it was all quite hard to understand, and of course may even have changed since.

So anyway, if you ask for format `XRGB8888` I think it may do what you want, and be displayable by the QTGL preview. Presumably you could also ask for 24-bit RGB, but then the QTGL preview wouldn't like it. I expect there might be other ways for you to call PIL directly and force it to do things differently...

Petros626, 2023-03-13:
_PIL_ is a fundamental part of your code as I see, so I just thought there's an option to save it as RGB with the QTGL preview. Switching to `cam.configure(cam.create_still_configuration()` without preview saves it as RGB. OpenCV also offers some functions for that.

To change the format here isn't recommended I guess:

https://github.com/raspberrypi/picamera2/blob/d07a49197622d636bd993a5bf1de321c259abf47/picamera2/picamera2.py#L619

davidplowman, 2023-03-13:
You're welcome to set whatever format you want when you create a configuration, though I'd probably leave the default choice there alone.

Petros626, 2023-03-13:
Yeah I'm thinking about to use a workaround and don't touch the source code. And as you said to change it to RGB would make problems with QTGL.

davidplowman, 2023-03-13:
That's true, though as I commented, by a slightly lucky coincidence it seems like you can use `XRGB8888` and get a 24-bit PNG without breaking the QTGL preview.

Petros626, 2023-03-13:
Oh, then I didn't understand you correctly. I'll see what is the best option in my case without changing the source code.

closed 2023-03-14 by Petros626