raspberrypi/picamera2 #620: Imaging through PyQT fails on switch mo

Naderschah, 2023-03-13:
**Please only report one bug per issue!**

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behaviour. Wherever possible, please include a small self-contained Python script that shows the bug.

**Expected behaviour**
A clear and concise description of what you expected to happen.

**Console Output, Screenshots**
If applicable, any console output or screenshots that show the problem and associated error messages.

**Hardware :**
 Please state what kind of Pi you are using, and what camera is attached.

**Additional context**
Add any other context about the problem here.

davidplowman, 2023-03-13:
Hi, am I right in thinking you're using an Arducam 64MP camera? I assume the problems are due to the very large amounts of memory that this camera may require, but I can't really confirm that as I don't have one here. You would probably need to talk to the module vendor for more support.

Sorry not to be more helpful. But we are hoping to make a release soon which uses less CMA memory, so maybe this will help you.

Naderschah, 2023-03-13:
Hey,
It is indeeed the 64MP camera.
I managed to get it working by using the cmd line implementation to take an image,
this makes me think that there is some link or something within the camera that doesnt get reset the way I remove the camera.
Would you happen to know how I can fully release the camera as happens when python does its cleanup (or what to look for in your source code to find it)?

Also would you happen to know if its theoretically possible to set cma to 1024? Everytime i try it assigns 300 something.

davidplowman, 2023-03-13:
The command line applications are able to use YUV420 rather than RGB, so the image buffers are much smaller. Unfortunately most Python modules don't support YUV420. You might consider capturing YUV420 instead and then using OpenCV to convert it to RGB. That would be a bit slow but the RGB buffer would be in user memory so wouldn't cause the same problems.

Calling `picam2.close()` should release everything. If it doesn't please send a minimal test script that demonstrates that the resources have not been freed, then we can take a look. Thanks.

So far as I know, 512MB is the largest for CMA.

Naderschah, 2023-03-13:
Ah alright, thank you for the explanation and alternatives.
I will retry my method original method on the upcoming releases

closed 2023-03-13 by Naderschah