raspberrypi/picamera2 #622: [HOW-TO] Fastest way to capture continuous full resolution .dng files

akarsh-sa, 2023-03-16:
Hello, 

I would like to know how to and how many raw images I can capture in .dng format (full resolution) per second. 

I have modified the [capture_dng.py](https://github.com/raspberrypi/picamera2/blob/main/examples/capture_dng.py) file by removing time.sleep(2), and I have looped the line: picam2.switch_mode_and_capture_file(capture_config, "full.dng", name="raw"). By running the script, I can capture 1 image every 2 seconds on average, that is 60 images in 2 minutes of total script runtime. Is there any other way to improve the capture speed?

I wondered if I could capture raw images in .dng format every 0.25 seconds for two minutes. Please let me know if my approach is wrong in fast saving the .dng files. 

My specs : Raspberry pi 4 model B (4 GB); camera module v3 NOiR

tvoverbeek, 2023-03-16:
The major part of the 2 seconds is the conversion of the raw buffer to a dng file by the pidng module.
If you want to speed it up you need to save buffers first for say 2 minutes and afterwards convert them to dng files.
On my Pi3b+ with camera module v3 NoIR  the conversion of a full resolution (4608x2592) takes almost 3 seconds (2975 msec).

davidplowman, 2023-03-20:
Hi, thanks for the question. I think there could be a number of different answers depending on which route you'd like to explore.

**Saving raw files quickly**

This does get asked from time to time because saving DNGs is not that quick. Some users have saved the raw buffers as (for example) bare numpy arrays, and then you can read them back and convert them later. It's worth noting that the exposure time and analogue gain need to be fixed, otherwise even raw files are likely to show some variation.

**Getting consistent processing**

It's possible to get 100% consistent processing in a raw and even in a JPEG file.

I'm going to assume you have a fixed environment. In this case I would work out what shutter time and gain you want, and program these manually. This will disable the AGC/AEC, which may otherwise make adaptive changes as time goes on. Your raw files should now be as "consistent" as possible (though see comments below for the imx708).

For JPEG files there are a few more things to fix (these changes have no effect on the raw files). I would decide what colour gains you actually want and program these manually too. This will stop the AWB from changing the colour balance subtlely over time. Actually the imx708 (camera v3) is a bit unusual in that the AWB _can_ affect raw files in a slightly indirect way, so I would certainly recommend fixing the colour gains even for an imx708 raw file (other cameras don't show this behaviour).

Next, in the camera tuning file you need to find the line that says `"ce_enable": 1,` and change it to `"ce_enable": 0,` and finally find the line that says `"n_iter": 100,` and change it to `"n_iter": 0,`.

Hope that helps, please get back to us if you need more information.

tvoverbeek, 2023-03-20:
@akarsh-sa I have been experimenting with the "saving raw buffers" way @davidplowman mentioned above.
If you want 2 minutes worth of 4 frames per second then you need to save 480 frames.
For the imx708 (as in the camera module v3) the frame size for raw can only be the size of one of the available 3 modes:
1536x864, 2304x1296, 4600x2592. SRGGB10_CSI2P stores 4 pixels in 5 bytes.
So for a single 1536*964 frame you need 1536x864x1.25 = 1,658,880 bytes. 480 frames need 796,262,400 bytes.
Moreover for each frame the metadata also has to be stored, since the save dng needs both.
Conclusion: only a Pi4b with 2GB might work,  probably best using a 4GB or 8GB model.
For 4600*2592 it s up to 14MB per frame. 480 frames is already 7GB.
Therefore I have used the smallest format (1536*864).
Here is the code for acquiring and storing 480 frames at 4 fps:
```
#!/usr/bin/python3

import pickle
import time
from picamera2 import Picamera2, Preview

picam2 = Picamera2()
picam2.start_preview(Preview.NULL)

# Specifying a small size in capture_config forces the smallest mode
capture_config = picam2.create_still_configuration(raw={"size": (640, 480)}, buffer_count=2)
picam2.configure(capture_config)
print(capture_config["raw"])

picam2.set_controls({"FrameRate": 4})

# Add other controls to force constant exposures here

time.sleep(2)

buffers = []
metadatas = []
starttime = time.monotonic()
while (time.monotonic() -  starttime) < 120.0:
    request = picam2.capture_request()
    buffers.append(request.make_buffer("raw"))
    metadatas.append(request.get_metadata())
    request.release()
endtime = time.monotonic()

print('Capture duration = ', endtime - starttime)
print(len(buffers))
print(type(buffers[0]))
print(type(buffers[0][0]))
print(len(metadatas))
print(metadatas[0])

starttime = time.monotonic()
# The dng conversion needs the used config, frames and metadatas
savedata=[capture_config["raw"], buffers, metadatas]
with open('testbuffers','wb') as f:
    pickle.dump(savedata, f)
endtime = time.monotonic()
print('pickletime = ', endtime-starttime)
```
My testbuffers file has size 798156718. for 481 frames.
Hre is the code to produce 481 dng files:
```
#!/usr/bin/python3

import pickle
from picamera2 import Picamera2

with open('testbuffers','rb') as f:
    data=pickle.load(f)

print(len(data))
config = data[0]
print(config)
buffers = data[1]
print(len(buffers[0]))
metadatas = data[2]
print(metadatas[0])

picam2 = Picamera2()
for i in range(len(buffers)):
   filename = 'full{0:03d}.dng'.format(i)
   print(filename)
   picam2.helpers.save_dng(buffers[i], metadatas[i], config, filename)
```
Each dng file has size 1659640

I have not tried to save the buffers directly to disk instead of to memory.
Might be possible with a fast SSD.

davidplowman, 2023-05-09:
Is there anything else to investigate here? If there are no further updates I'll consider closing this issue shortly. Thanks!