raspberrypi/picamera2 #624: overlay file system is enabled, why picamera2 no work?

yq3w, 2023-03-17:
Due to a power outage that has damaged the raspberry pie zero 2w system, I want to use the overlay file system to protect the system security. Picamera2 can be used for normal photography on the raspberry pie distribution system. After opening the overlay file system in raspi-config, it is no longer possible to take full resolution photos. Is there a way to add swap areas, caches, and the like to make Picamera2 work properly?

pi@raspberrypi:~ $ cat /proc/cpuinfo | grep Revision | awk '{print substr($NF,length($NF)-5,6)}'
902120
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.15.84-v7+ #1613 SMP Thu Jan 5 11:59:48 GMT 2023 armv7l GNU/Linux
pi@raspberrypi:~ $ head -1 /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"

yq3w, 2023-03-17:
pi@raspberrypi:~ $ python3 capture_jpeg.py
[0:01:17.535504083] [882] INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:01:17.634467781] [883] INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media0 and ISP device /dev/media1
[0:01:17.650710646] [882] INFO Camera camera.cpp:1028 configuring streams: (0) 4056x3040-BGR888
[0:01:17.651476114] [883] INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 4056x3040-SBGGR12_1X12 - Selected unicam format: 4056x3040-pB CC
Killed

davidplowman, 2023-03-17:
Hi, I've tried this and it seemed to work for me. Perhaps you can explain a bit more what you did and what happened. Here's what I tried:

1. It seems to me that you are using the 32-bit Lite version of the OS, is that correct? So I flashed that onto an SD card.
2. After the Pi had booted I ran `sudo apt update` and `sudo apt upgrade`.
3. Then I enabled the overlay file system. I left the boot partition writable (which seems to be the default).
4. After rebooting again, I started Picamera2 and took full resolution capture.

I note that you used the `capture_jpeg.py` script. I wouldn't expect this to work as it creates a Qt preview window. Though it wouldn't have worked before you created the overlay file system either. Or have you installed X Windows and Qt on top of your OS Lite?

If you could supply more information that would be helpful. Things like: 

* Did you install the Lite OS, or the desktop one? If the Lite OS, did you install X Windows and Qt on top?
* Are you trying this on a faily "clean" image, or have a large number of other changes been made?
* Are you able to create a preview window (which would normally be a DRM overlay on the Lite OS)?
* Does the camera start?
* At what point does it fail - is it when you write to the file or before?
* Can you capture full resolution images if you don't write them to a file (e.g. using `capture_array`)? Is other software able to write to the file system?

Though I should add that I was using a Pi 3 as I don't have a Pi Zero 2W available at the moment. But I can try that if the answers to the above questions don't help. Thanks.

yq3w, 2023-03-17:
Using the libcamera-still -o a.jpg command can successfully take 4056 resolution photos

davidplowman, 2023-03-17:
Thanks for the update. I've reproduced the problem by setting `total_mem=512` in my `/boot/config.txt`.

It does indeed seem like memory is the problem. I'm guessing that the overlay filesystem doesn't work with the system swap file which is why this has started happening.

`libcamera-still` is more efficient in terms of memory usage because it uses YUV420 buffers, so a 12MP image is "only" 18MB. Unfortunately Python modules generally have no support for YUV420, so buffers all have to be 24-bit RGB which are an enormous 36MB each.

In Picamera2 you could set the format to "YUV420", but the catch is then that there are no easy functions to save the images as JPEG files! 

So anyway, if you want advice on the swap file issue then you probably need to ask on a more general Raspberry Pi forum - it's not something I know anything about and it's not something I can fix in Picamera2.

As for workarounds, I don't know if `libcamera-still` is enough. Otherwise you could capture an array in the YUV420 format, close the camera object, and then convert the YUV420 image to RGB and save it. But that's quite ugly.

Some good news is that we do have some fairly large memory improvements coming in Picamera2 in the next couple of weeks, and these may be enough to make this use case work (though it is of course hard to be certain).