raspberrypi/picamera2 #625: [HOW-TO] prevent image auto adjusting

vb216, 2023-03-18:
**Please only ask one question per issue!**

**Describe what it is that you want to accomplish**

I'm using a new v3 camera module in a small film scanner unit, and seeing something I don't quite understand, where each frame has some sort of adjustment going on. I think its focal, but as far as I can tell I've set to manual focus.  Even if the image is out of focus, I would prefer it to stay as-is, so that I can adjust the settings on the 20x magnification lens or the physical position of the camera itself.

**Describe alternatives you've considered**

I've looked through the manual PDF and tried tuning a few settings, and checked that the focus value isn't changing somehow, but the problem persists. 

I'm using this little test:

```
from picamera2 import Picamera2, Preview
from libcamera import controls
import time
from time import sleep

Picamera2.set_logging(Picamera2.DEBUG)
picam2 = Picamera2()
picam2.start_preview(Preview.QT, x=100, y=200, width=1024, height=768)
picam2.start()
picam2.set_controls(
    {
        # controls for main frame: disable all regulations
        "AeEnable": False,  # AEC/AGC algorithm
        # disable noise reduction in main frame because it eats stars
        "NoiseReductionMode": controls.draft.NoiseReductionModeEnum.Off,
        # disable automatic white balance algorithm and set colour gains manually
        "AwbEnable": False,
        "AfMode": controls.AfModeEnum.Manual, "LensPosition": 10.0,
    }
)

while True:
  sleep(1)
  print(picam2.capture_metadata()['LensPosition'])

```

The while loop was just to check the position (which remains at 10), same issue without the loop. Issue remains with different lens position values.  Camera seems fine for normal 'long range' type shots. 

**Additional context**

It seems each frame the image is different, perhaps these images help:

[image]

[image]

I don't think its related, but I seem to get a stack trace too, but I think this is just on the stream closing by the looks of it:

```
[Sat Mar 18 11:41:03 2023] Call trace:
[Sat Mar 18 11:41:03 2023]  __media_pipeline_stop+0x8c/0xa0 [mc]
[Sat Mar 18 11:41:03 2023]  media_pipeline_stop+0x34/0x50 [mc]
[Sat Mar 18 11:41:03 2023]  unicam_stop_streaming+0x78/0x170 [bcm2835_unicam]
[Sat Mar 18 11:41:03 2023]  __vb2_queue_cancel+0x3c/0x220 [videobuf2_common]
[Sat Mar 18 11:41:03 2023]  vb2_core_queue_release+0x2c/0x58 [videobuf2_common]
[Sat Mar 18 11:41:03 2023]  _vb2_fop_release+0x88/0xa8 [videobuf2_v4l2]
[Sat Mar 18 11:41:03 2023]  unicam_v4l2_release+0x58/0xe0 [bcm2835_unicam]
[Sat Mar 18 11:41:03 2023]  v4l2_release+0xc8/0x108 [videodev]
[Sat Mar 18 11:41:03 2023]  __fput+0x78/0x258
[Sat Mar 18 11:41:03 2023]  ____fput+0x18/0x28
[Sat Mar 18 11:41:03 2023]  task_work_run+0x88/0xe0
[Sat Mar 18 11:41:03 2023]  do_exit+0x320/0xa90
[Sat Mar 18 11:41:03 2023]  do_group_exit+0x3c/0x98
[Sat Mar 18 11:41:03 2023]  get_signal+0x87c/0x8d0
[Sat Mar 18 11:41:03 2023]  do_notify_resume+0x30c/0x448
[Sat Mar 18 11:41:03 2023]  el0_svc+0x80/0x88
[Sat Mar 18 11:41:03 2023]  el0t_64_sync_handler+0x98/0xc0
[Sat Mar 18 11:41:03 2023]  el0t_64_sync+0x18c/0x190
[Sat Mar 18 11:41:03 2023] ---[ end trace 0000000000000000 ]---
```

tvoverbeek, 2023-03-18:
In your set_controls you fix a lot of settings.
Have you tried to fix the ExposureTime?

vb216, 2023-03-18:
Hi - thanks for the suggestion, unfortunately doesn't seem to have cracked it. I saw in the exposure_fixed example that analog gain was set too, so I tried adding:

"ExposureTime": 4000,
"AnalogueGain": 1.0,

(as well as without AnalogueGain being set) and issues persists same as before

vb216, 2023-03-19:
Ah.. I might have found it. 
I was trying libcamera-hello locally with a screen plugged in, and noticed the image wobble was alot quicker. I've got a fan mounted on the side of the case to keep the equipment inside, and it looks like this is causing some resonance. If I dampen things down with my hand, it seems alot better.

closed 2023-03-19 by vb216