raspberrypi/picamera2 #626: [HOW-TO] apply NoiseReductionMode

Petros626, 2023-03-19:
Hey,

I've seen that one user already asks for the correct use of the `NoiseReductionMode` (https://github.com/raspberrypi/picamera2/issues/468). 
In my script it seems optically (from previewWindow), that it doesn't take any effect. 
Normally after calling the script the `tuning_file`, `options` and `controls` should be set one time and be applied as long as the script is running.

```python
class ImageTaker:
    tuning_file = Picamera2.load_tuning_file("ov5647_custom.json")
    picam2 = Picamera2(tuning=tuning_file)
    picam2.options["quality"] = 95 # best quality
    picam2.options["compress_level"] = 0 # no compression
    picam2.set_controls({"NoiseReductionMode": controls.draft.NoiseReductionModeEnum.HighQuality})

    def __init__(self, imgdir, res):
        self.res = res
        self.imgdir = imgdir
        self.imgnum = 1
        self.key_flag = 0
        self.dirpath = None
        self.initialize_camera()
```

Petros626, 2023-03-20:
solved in https://github.com/raspberrypi/picamera2/issues/468

closed 2023-03-20 by Petros626

davidplowman, 2023-03-20:
Just to add that the effects of changing the noise reduction can be fairly subtle, and it of course depends how noisy the image is to begin with.  Note also that if you switch camera mode (for example `switch_mode_and_capture_file`) then it may choose a new default denoise mode for the capture. config To avoid that, always specify the noise reduction mode that you want when you create the mode, for example
```
# If you don't specify the noise reduction mode, it will default back to HighQuality
capture_config = picam2.create_still_configuration(controls={'NoiseReductionMode': controls.draft.NoiseReductionMode.Fast})
...
picam2.switch_mode_and_capture_file(capture_config, "test.jpg"))
```

Petros626, 2023-03-20:
Thanks I now know how to use it, was just a syntax issue.

davidplowman, 2023-03-20:
Great, glad you've sorted it out. If there's anything else - you know where we are!!