raspberrypi/picamera2 #627: [BUG] INFO Camera camera.cpp:841 Pipeline handler in use by another process

jatdow, 2023-03-19:
I saw this issue reported previously, but it was fixed by reinstalling picamera2. I tried this , but without success. So anyway: Rpi 4B, with picamera 3. Here's the program, which is pretty basic, I think:
[image]
and here's the traceback:
[image]
It works SOMETIMES, but usually winds up failing like this.
Any ideas please?

tvoverbeek, 2023-03-19:
You are starting the camera twice, once in picam2.start and the second time in picam2.start_and_capture_file.
To take a picture start_and_capture_file i sufficient. See section 6.6 in the manual.

davidplowman, 2023-03-20:
Yes, thanks for the explanation @tvoverbeek !

@jatdow  I'll assume this one is solved now, but please get back to us if there's anything else. Thanks!

davidplowman, 2023-03-20:
Hi, that little 3 line script works for me. From the error messages, it looks as though one of your other processes is still lurking in the background and is holding on to the camera so you now can no longer acquire it for use. Please see if you can find and kill those processes and then I think things should return to normal.

jatdow, 2023-03-21:
I'll keep looking for parasitic processes, but I'm not convinced. This is a clean install of Raspian for this specific job, and the problem is intermittent. I got a few hours of it working last night as a cron job, then not working this morning. It also works better with a screen attached than headless. I'd wondered about path/environment under different regimes. Frustrating!

davidplowman, 2023-03-21:
Yes, I think people do generally have permissions and path problems when running stuff through cron jobs, or automatically at startup. If libcamera can't acquire a resource it's possible that you get "resource busy" messages when "resource not permitted" would be more accurate.

jatdow, 2023-03-23:
For future folk, the resolution was triggered by a throwaway line in the manual. The camera actually requires a preview, because the preview feeds the picture acquisition. My system would run for a while, but would lock shortly after I disconnected the monitor to go headless (security cam intention), so I guessed the picam2 software would hang if it couldn't find a monitor to display a preview. The solution was to include a line such as picam2.start_preview(Preview=NULL). So my prediction is that the slimmed down code Picamera2.start_and_capture_file() won't work headless.

davidplowman, 2023-03-23:
Thanks for looking into this. Are we saying that, if you start the camera with a preview window, and then unplug the monitor, then it will stop running? We could include a note to that effect in the documentation, though I'm not sure it's a behaviour that we'd change.

jatdow, 2023-03-23:
That's what happened for me. So headless operation may be an issue. But maybe other folks' mileage may vary.

closed 2023-03-23 by jatdow