raspberrypi/picamera2 #629: [HOW-TO] pass a dictionary to function parameters

Petros626, 2023-03-20:
Hey,

Im trying to pass my config dict to the function: https://github.com/raspberrypi/picamera2/blob/d07a49197622d636bd993a5bf1de321c259abf47/picamera2/request.py#L182

The problem is, that it doesn't work with the ton of combinations I tried. 
The only working way is: 
```python 
request.make_array("main")
```
Why not 
```python
request.make_array("main", config["format"])
or
request.make_array("main", config["format"], config["size"], config["stride"])
``` 
with a dict like `config = {"format": "XRGB8888", "size": "1080,1920", "stride": "7680"}`. Unfortunately the documentation under point 6.3/6.4 didn't help me.

tvoverbeek, 2023-03-20:
As written in section 6.4 the request is the data returned from the camera. You cannot configure the camera afterwards when the request has already been returned to picamera2. Configure the camera the normal way before capturing a request.
The returned request contains all configured streams, hence in make_array you have to specify which stream you want, hece the request.make_array("main") function.
make array converts the returned buffer (one dimensional array with packed pixel data)  to a 2 dimensional arrray with unpacked pixel data.

Petros626, 2023-03-20:
Unfortunately, it is not the case that previously defined configurations are adopted without any problems. 
```python
picam2.configure(self.picam2.create_preview_configuration(main={"format": "XRGB8888", "size": (imgW, imgH)}))
```
I expected a 3-channel RGB array, but when I call `request.make_array("main")` I get a 4-channel RGBA array, thats my point. Another author explained me here (https://github.com/raspberrypi/picamera2/issues/619#issuecomment-1466087978) about the channels with `PIL` library.

Petros626, 2023-03-27:
@davidplowman  Hey could you take a lot for the array format problem.

davidplowman, 2023-03-27:
> Unfortunately, it is not the case that previously defined configurations are adopted without any problems.
> 
> ```python
> picam2.configure(self.picam2.create_preview_configuration(main={"format": "XRGB8888", "size": (imgW, imgH)}))
> ```
> 
> I expected a 3-channel RGB array, but when I call `request.make_array("main")` I get a 4-channel RGBA array, thats my point. Another author explained me here ([#619 (comment)](https://github.com/raspberrypi/picamera2/issues/619#issuecomment-1466087978)) about the channels with `PIL` library.

Sorry, I didn't seem to spot this issue earlier.

I'm not quite clear what the problem is here - the requested format is `XRGB8888` so you'll get back a 32-bit value (or 4 8-bit values) for each pixel, one of them being the "X" value (which of course looks much like an RGBA 32-bit pixel). Have I misunderstood what's happening, or are you able to add further explanation? Thanks.

Petros626, 2023-03-28:
The problem is, that in #619 you mentioned when I'm using `XRGB8888` _PIL_ will save it as `RGB` and not `RGBA`. And I expected when I configure the preview with this format and use the function `make_array`the `RGB` format will be maintained.

preview: `XRGB8888` -> save result: `RGB `image (works)
use function: `make_array("main")` -> save result `RGBA` (this isn't what I want)

Petros626, 2023-03-28:
then I have now understood how it is meant. You are right, I can't get a correct format at all, because the format change is caused by _PIL_ when saving. If I want a different format, I have to make a conversion myself after calling the function.

closed 2023-03-28 by Petros626