raspberrypi/picamera2 #631: [SUGGESTION]add Exif metadata tag for picamera2 API.

vignesh341, 2023-03-22:
**Please only include one suggestion or features request per issue!**

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

vignesh341, 2023-03-23:
Hi! 
    thanks for your reply, Here I list the exif tags which I tried with picamera2's Piexif  that doesn't works 
=> IFD0.Make
=> IFD0.Model
=> IFD0.Software
=> IFD0.Copyright

this are the tag I were used in the previous version of the picamera API

davidplowman, 2023-03-24:
Hi, I tried running exiftool on one of the captured images, and I see
```
  + [IFD0 directory with 4 entries]
  | 0)  Make = Raspberry Pi
  | 1)  Model = /base/soc/i2c0mux/i2c@1/imx708@1a
  | 2)  Software = Picamera2
```
It's true that I don't see a copyright field. What are you expecting to see there, or are you asking to be able to specify something to put in there? I also haven't understood why you don't see these other fields - how are you capturing your image? Thanks!

vignesh341, 2023-03-25:
Hi, Thanks for the reply.
I am sorry if I confused you a little bit. The tags you displayed were the exiftool default output of a captured image, but I had found that those tags could be changed in an earlier version of Picamera. But I did not discover that this tag was not functioning in the Picamera2 api.

davidplowman, 2023-03-27:
Hi again, so is your request that all the fields (Make, Model, Software and Copyright) are assigned default values (as most of them are now), but that you could override them all with your own custom values if you want? Thanks.