raspberrypi/picamera2 #636: [HOW-TO] Camera Calibration (undistort lense) with OpenCV use in PiCamera2

Petros626, 2023-03-24:
Hello,

Since I found no lens correction for the Raspberry Pi cameras in the documentation (https://datasheets.raspberrypi.com/camera/raspberry-pi-camera-guide.pdf), as well as for other manufacturers, I wanted to create an interface or connection between these two libraries myself. I imagined this by means of array and array transformation. Thus, users can not only load a tuning_file, but also be sure that the distortions are removed  

**Visualised:**
Camera Calibration (_OpenCV_)  **->**  save _.xml_ / _.yaml_ file with parameters  **->**  load/bind in _PiCamera2_ code additional to _tuning_file_  

**Approach/Consideration:**
to be able to apply the parameters, the function `cv2.undistort()` would be the easier alternative than using `cv2.getOptimalNewCameraMatrix()` in relation to the return parameters, as input and output array can be worked with there.

Is there a simple way to realise this connection, using arrays for a stream or image capture. As always, the code is published as an example.

**Docs:**
cv2.undistort():
https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga69f2545a8b62a6b0fc2ee060dc30559d
cv2.getOptimalNewCameraMatrix():
`(https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga7a6c4e032c97f03ba747966e6ad862b1)

davidplowman, 2023-03-24:
Hi, this is an interesting topic, though not one I've spent any time on myself. I would be very happy to publish some example code showing how to do this sort of thing. I also wonder whether this kind of thing could be done efficiently with the GPU, though that's a whole other question!

Petros626, 2023-03-25:
I will try to work on something that could be a first solution, but I am not sure if swapping arrays is enough.

davidplowman, 2023-04-11:
I'll close this issue for now, if that's ok, as I don't think there's anything specific for us to investigate at the moment. But please do open another issue if something else comes up. Thanks!

closed 2023-04-11 by davidplowman

Petros626, 2023-04-20:
@davidplowman Hey I worked on a solution, which can load the .json `tuning_file` and the .json `calibration file` for the camera with the matrix (`mtx`) and distortion (`dst`). The preview stream seems slowed down, I don't know what exactly causes it, but maybe you have an idea? I used `capture_array` instead of the `request`, tested both and it seems similiar for me.

```python

from picamera2 import Picamera2
from argparse import ArgumentParser
from keyboard import is_pressed
from termios import tcflush, TCIOFLUSH
from sys import stdin, exit
from libcamera import controls
from json import load
from numpy import array, float64
from cv2 import (
    cvtColor, COLOR_RGBA2RGB, undistort, imshow, getOptimalNewCameraMatrix, destroyAllWindows,
    namedWindow, WINDOW_NORMAL, startWindowThread, resizeWindow, moveWindow
)

parser = ArgumentParser()
parser.add_argument("--res", help = "Required resolution in WxH. To avoid erros find out about the supported resolutions of your camera model.",
                       default = "1920x1080")
args = parser.parse_args()

if not "x" in args.res:
    print("Specify resolution with x as WxH. (Example: 1920x1080).")
    exit()
imgW, imgH = map(int, args.res.split("x"))

tuning_file = Picamera2.load_tuning_file("ov5647_custom.json")
picam2 = Picamera2(tuning=tuning_file)
# Set options for saving images
picam2.options["quality"] = 95 # best quality
picam2.options["compress_level"] = 0 # no compression

picam2.configure(picam2.create_preview_configuration(main={"format": "XRGB8888", "size": (1920, 1080)}, controls={"NoiseReductionMode":controls.draft.NoiseReductionModeEnum.HighQuality}))
picam2.start()

winname = "Calibrated (undistorted) Image taker"
namedWindow(winname, WINDOW_NORMAL)
resizeWindow(winname, 1000, 900)
moveWindow(winname, 915, 72)
startWindowThread()
key_flag = 0

with open("calibrate_camera.json", "r") as f:
    calibration_file = load(f)
mtx = array(calibration_file["mtx"], dtype=float64)
dist = array(calibration_file["dist"], dtype=float64)

try:
    while 1:
        if key_flag == 0:
            key_flag = 1
            # request - faster?
            #request = picam2.capture_request()
            #array = request.make_array("main")
            #2 direct capture -faster ?
            array = picam2.capture_array("main")
            new_cv_img = cvtColor(array, COLOR_RGBA2RGB)
            h, w = new_cv_img.shape[:2]
            optimal_camera_matrix, roi = getOptimalNewCameraMatrix(mtx, dist, (w, h), 0, (w, h))
            dst = undistort(new_cv_img, mtx, dist, None, optimal_camera_matrix)
            imshow(winname, dst)
        else:
            key_flag = 0
        #request.release() 
        if is_pressed("q"):
            print("\r++++++++++++++++++++++++++++++++++++++++++++++")
            print("\rInterruption: stop preview and close camera...")
            break      
finally:
    destroyAllWindows()
    picam2.stop()
    picam2.close()
    tcflush(stdin, TCIOFLUSH)
```

davidplowman, 2023-04-24:
Hi, thanks for posting that, it looks interesting.

One obvious thing to do would be to change the format from `XRGB8888` to `RGB888`, that would presumably save you doing the RGBA to RGB conversion.

After that I'm not sure. Have you profiled the undistort function to see how long it's taking? I'm also aware that imshow  can be relatively costly for large images, but getting the built-in hardware accelerated preview to display those images is not so easy either.

As you've noticed, copying the images with capture_array is often not as harmful as you might expect. In fact it is frequently helpful. That's because the buffers that the hardware fills are not cached, so accessing pixels in there using anything other than a simple memcpy can become very slow.

Petros626, 2023-04-24:
Hey,
the colour format can be changed to `RGB888` thats correct, but when I use `request.save` or `capture.save` function (what I intend to do) we know from here: https://github.com/raspberrypi/picamera2/issues/619#issuecomment-1466087978, that a little workaround is necessary.

I checked it with `cProfile` and it jumps between **0.12x** and **0.13x** seconds. Fir the preview window, I thought the window thread could be helpful, but generally the stream is slowed down.